ethz-asl/rovio #113: A question about NormalVectorElement

shishiteng, 2017-02-14:
I spent a lot of time to read the code but did not understand what does getM and getN mean in LWF::NormalVectorElement.

`Eigen::Matrix<double,3,2> getM() const {		
    Eigen::Matrix<double,3,2> M;		
    M.col(0) = -getPerp2();		
    M.col(1) = getPerp1();		
    return M;		
  }		
  Eigen::Matrix<double,3,2> getN() const {		
    Eigen::Matrix<double,3,2> M;		
    M.col(0) = getPerp1();		
    M.col(1) = getPerp2();		
    return M;		
  }`

Can anyone help me?

nhanph, 2017-02-24:
From my understanding, they are the two vectors projected from the 3D vector to the tangential space around the bearing vector.

bloesch, 2017-03-12:
what @nhanph says is correct. getPerp1 and getPerp2 basically span the tangential space. the maths behind are not that easy, especially when it comes to derivatives. a paper is under submission which provides some more details. it will get linked into the readme as soon as it is published.

shishiteng, 2017-03-29:
Thank for reply @bloesch @nhanph ,I‘m looking forward to your new paper.

closed 2017-03-30 by bloesch