ethz-asl/rovio #117: process has died

juwikuang, 2017-03-21:
when I run "roslaunch rovio_rosbag_node.launch ", the two windows disappears immediately, throwing the following exception:

process has died

I do not have an issue running "roslaunch rovio_node.launch". I saw the 3D screen with 3 axes.
...................error details..................

eric@eric-Spin-SP315-51:~/catkin_ws/src/rovio/launch$ roslaunch rovio_rosbag_node.launch 
... logging to /home/eric/.ros/log/a1398f00-0e08-11e7-a232-000ec6cf37c6/roslaunch-eric-Spin-SP315-51-16979.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://eric-Spin-SP315-51:38205/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /rovio/cam0_topic_name: /cam0/image_raw
 * /rovio/cam1_topic_name: /cam1/image_raw
 * /rovio/camera0_config: /home/eric/catkin...
 * /rovio/camera1_config: /home/eric/catkin...
 * /rovio/filter_config: /home/eric/catkin...
 * /rovio/imu_topic_name: /imu0
 * /rovio/rosbag_filename: /home/eric/catkin...

NODES
  /
    rovio (rovio/rovio_rosbag_loader)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rovio-1]: started with pid [16997]
Set Camera Matrix to:
458.654       0 367.215
      0 457.296 248.375
      0       0       1
Set distortion parameters (Radtan) to: k1(-0.283408), k2(0.0739591), k3(0), p1(0.00019359), p2(1.76187e-05)
Testing Prediction
==== Test successful (8.23703e-08) ====
==== Test successful (5.9316e-08) ====
Testing cameraOutputCF
==== Test successful (4.44089e-08) ====
Testing imuOutputCF
==== Test successful (7.9955e-09) ====
Testing attitudeToYprCF
==== Test successful (4.6604e-08) ====
Testing transformFeatureOutputCT
==== Test successful (1.0206e-08) ====
Testing LandmarkOutputImuCT
==== Test successful (3.57083e-08) ====
Testing FeatureOutputReadableCT
==== Test successful (1.21661e-08) ====
Testing pixelOutputCT (can sometimes exhibit large absolut errors due to the float precision)
==== Model jacInput Test failed: 74.0615 is larger than 1 at row 1(def_0) and col 0() ====
  -72871.9  -72797.9
Testing zero velocity update
==== Test successful (3.04311e-10) ====
==== Test successful (1.39778e-10) ====
Recording
Storing output to: /home/eric/catkin_ws/src/rovio/launch/rovio/2017-03-21-09-26-18_25_4_6_1_0.bag
-- Filter: Initializing using accel. measurement ...
-- Filter: Initialized at t = 1403637130.5
init done
[rovio-1] process has died [pid 16997, exit code -11, cmd /home/eric/catkin_ws/devel/lib/rovio/rovio_rosbag_loader __name:=rovio __log:=/home/eric/.ros/log/a1398f00-0e08-11e7-a232-000ec6cf37c6/rovio-1.log].
log file: /home/eric/.ros/log/a1398f00-0e08-11e7-a232-000ec6cf37c6/rovio-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
eric@eric-Spin-SP315-51:~/catkin_ws/src/rovio/launch$ 
--------------------------------------------------------------------------------------------------------------------

juwikuang, 2017-03-21:
I also post my steps of installation here:

Steps

1.Installed Ubuntu 16.04

2.Installed ROS kinetic

	Followed this instruction: http://wiki.ros.org/kinetic/Installation/Ubuntu
	In 1.4 chose the recommended approach: Desktop-Full Install: (Recommended)

3.Installed kindr(https://github.com/ethz-asl/kindr)

	Tried (installing from packages, failed) “sudo apt-get install ros-indigo-kindr-*” failed. 

error massage:
==========================================================
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-indigo-kindr-*
E: Couldn't find any package by glob 'ros-indigo-kindr-*'
E: Couldn't find any package by regex 'ros-indigo-kindr-*'
===========================================================
 
	So, I Installed it with cmake.

        In hindsight, I think I should have installed ROS-indigo instead of ROS-Kinetic

4.Installed lightweight_filtering

	I have to clone it the ROVIO from https://github.com/ethz-asl/rovio.git so that I can use the command "git submodule update --init --recursive"

5.Installed catkin-tools (http://catkin-tools.readthedocs.io/en/latest/installing.html)

6.try "catkin build" 
	http://catkin-tools.readthedocs.io/en/latest/verbs/catkin_build.html

7.Install without opengl scene -- Execute "catkin build rovio --cmake-args -DCMAKE_BUILD_TYPE=Release -DMAKE_SCENE=ON"
	put rovio folder to catkin workspace, build
	with error:

==========================================error massage start====================================
_______________________________________________________________________________
Errors     << rovio:cmake /home/eric/catkin_ws/logs/rovio/build.cmake.001.log  
CMake Error at /home/eric/catkin_ws/src/catkin/cmake/catkin_package.cmake:298 (message):
  catkin_package() absolute include dir
  '/home/eric/catkin_ws/devel/share/xmlrpcpp/cmake/../../../include/xmlrpcpp'
  does not exist
Call Stack (most recent call first):
  /home/eric/catkin_ws/src/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:82 (catkin_package)

cd /home/eric/catkin_ws/build/rovio; catkin build --get-env rovio | catkin env -si  /usr/bin/cmake /home/eric/catkin_ws/src/rovio --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/eric/catkin_ws/devel/.private/rovio -DCMAKE_INSTALL_PREFIX=/home/eric/catkin_ws/install -DCMAKE_BUILD_TYPE=Release; cd -
...............................................................................

============================================error massage end===================================

	Copied the xmlrpcpp folder from the shared folder, run again, successed.

8. Install with opengl scene
	just follow the instruction, successed.

9. Source the bash
	because of the error: [rospack] Error: package 'rovio' not found
	Googled it found the solution : http://stackoverflow.com/questions/27053334/ros-package-not-found-after-catkin-make

	# ROS settings
	source <your_workspace_path>/catkin/devel/setup.bash

	I have to repeat step 9 every time I restart the computer.

10. Create a rovio folder at bag folder 
reason: https://github.com/ethz-asl/rovio/issues/72

bloesch, 2017-03-22:
Two first thoughts:
- could you please test it with the euroc datasets?
- please check you have enough diskspace

juwikuang, 2017-03-22:
yes, i used euroc datasets. and i have enough diskspace (200G)

bloesch, 2017-03-22:
then maybe it is ROS-indigo which is causing the problem.
you can also build in debug mode to get more information.
also: do you install with or without the opengl scene? your comments are not clear above.

juwikuang, 2017-03-22:
i installed ROS-kinetics, i installed with opengl scene. thanks. I think maybe i should install ROS-indigo and I am installing it.

bloesch, 2017-03-22:
sry, I meant ROS-kinetic...

juwikuang, 2017-03-22:
thanks, i am re-installing everything. will let you know the results.

juwikuang, 2017-03-27:
i reinstalled from scrach with ubuntu 14.04 and ros-indigo. the problem solved.

closed 2017-03-27 by juwikuang