ethz-asl/rovio #120: rovio run with laptop camera or usb camera issues

yujianyuan, 2017-04-19:
hi,everyone,thanks for reading,
i have build rovio by:`$catkin build rovio --cmake-args -DCMAKE_BUILD_TYPE=Release`
and then run rovio by:`$rosrun rovio rovio_node ` `rosbag play /media/MH_01_easy.bag`
then it seems runs ok except **some error: no valid coordinate data!**
[image]

i want to run rovio with my laptop camera and one arduino IMU, 
but i don't no **how to connect rovio with the camera and the IMU**, 

by rostopic list i can find 
```
/abss/twist
/cam0/image_raw
/cam1/image_raw
/clock
/imu0
/leica/position
/odometry
/pose
/rosout
/rosout_agg
/rovio/T_G_W
/rovio/extrinsics0
/rovio/imu_biases
/rovio/markers
/rovio/odometry
/rovio/patch
/rovio/pcl
/rovio/pose_with_covariance_stamped
/rovio/transform
/tf

```
**do i need to public three rostopic: /cam0/image_raw /cam1/image_raw /imu0 to rosmaster?
if so which data struct i need to feed to the rostopics?**

Any one knows anything is fine,thanks a lot !

marco-tranzatto, 2017-04-20:
Hi,

Did you have a look at the webpage about camera and IMU synchronization linked in the Wikipage of this repo?
Here's the link: [http://grauonline.de/wordpress/?page_id=1951](http://grauonline.de/wordpress/?page_id=1951)

yujianyuan, 2017-04-20:
Thanks a lot ! I have read the wiki link ,it helps a lot ! I will have a try. 
so I just need to trigger the camera by the Arduino board ,and then public IMU data and image data at the same time?
Is there any other document about the data structure of the IMU data and image data?
Thanks so much !

juwikuang, 2017-05-05:
@yujianyuan I wrote some code for using laptop camera and sends ROS messages, hope that would be useful.

https://github.com/juwikuang/camera_message

For IMU data from usb port:

https://github.com/juwikuang/usb_sensor_message

closed 2018-01-11 by bloesch