ethz-asl/rovio #121: ERROR: No valid coordinate data!

juwikuang, 2017-05-05:
[image]

I have installed ROVIO. After that, I run it with several datasets including MH01,02,03,04 and Vicon01. I got the same error each time I ran it. And the coordinators in the Scene window drift away very soon.

I thought it was caused by using only one camera so I modified the CMakeList.txt file setting ROVIO_NCAM = 2, that did not help.

I then tried building it without scene, no help.

I also tried setting doVECalibration = false. no help.

I noticed that at the beginning it was very slow. if I hit space and pause the bag, the tracker0 window take 10 seconds to pause. The scene window updates every 5 seconds, and if I try to move it by dragging, it takes 5 seconds, too. I am not sure if the issue was cause by my slow computer.

The algorithm keeps getting new features even when the helicopter is not moving. I see blue numbers keep increase. I also see features being removed and adding in in verbose mode.

Do I need to calibrate the camera and IMU sensors before I use ROVIO with the provided datasets?

juwikuang, 2017-05-05:
[image]

bloeschm, 2017-05-05:
have you compiled in release mode? in debug mode rovio is too slow and will soon miss up (I think due to skipping measurements)

juwikuang, 2017-05-05:
yes, I compiled in release mode. I used the command in the README.md file.

bloesch, 2017-05-05:
Does the tracking fail since the beginning?

juwikuang, 2017-05-05:
yes, it gives the error after five seconds.

bloesch, 2017-05-05:
what if you run the rosbag very slowly? also, have you tried using the rovio_rosbag_node.launch? there you can directly specify the rosbag to be loaded which is then directly loaded by rovio such that no messages are missed.

juwikuang, 2017-05-05:
thanks, it helped in some degree.

i tried running it slow by using --rate 0.1, no help.

I then tried rovio_rosbag_node.launch. it looks fine. the numbers now are green and they do not change. the problem is that I do not get a scene window and I have no idea if it drifts or not.

I deleted everything in the build and devel folder and rebuilt the package with 

`catkin build rovio --cmake-args -DCMAKE_BUILD_TYPE=Release -DMAKE_SCENE=ON`

i still dont have the scene window.

I run the rovio_node.launch , but the scene window seems not response to the ros bag.

What should i do to see the scene window? thanks.

bloesch, 2017-05-05:
The scene window should be visible when using -DMAKE_SCENE=ON with rovio_node.launch. Also, be sure that when changing the cmake parameters you properly rebuild your workspace.

juwikuang, 2017-05-07:
@bloesch , I ran rovio_node again just now and everything is fine. I no longer get "ERROR: No valid coordinate data!" Thanks very much and sorry for waiting your time.

juwikuang, 2017-05-07:
I am guessing it was caused by the poor computer. too slow it the computer.

closed 2017-05-07 by juwikuang

raoulj, 2018-01-10:
Setting `--rate 0.1` as  @bloesch suggested fixed the drift issue for me.

hook25127, 2020-04-07:
@juwikuang 
Hello, 
I got the same error: "ERROR: No valid coordinate data!"
How you running it slow by using **--rate 0.1**?