ethz-asl/rovio #122: how can improve rovio with quick motion ,especially about rotation

xiaopenghuang, 2017-05-12:
When i rotate my device , if the speed is not fast , rovio can give me a not bad odometry compared with motion cap system, if the speed is a little fast , the odometry always not good, which i think the speed is not 
good estimated, the speed is too little or too big, if the speed more fast , then speed quick diverged, the odometry is drift quickly. And imu is mti-g-700. 
 How can i improve this situation.Is it right , if you move fast ,the feature you tracks is not tracked well ,always create some new.The image update process did not work.
Anyone has any suggestion about it .Thanks.

ParthaEth, 2017-06-12:
@xiaopenghuang any improvements? I see this problem too. If the motion is too slow then sometimes ROVIO turns in the wrong direction(Right instead of left for example.) If yes what is the solution?

bloesch, 2017-06-12:
@xiaopenghuang: it depends what you mean by fast? Rovio should be able to handle fast rotations as long as they are only temporary and do not lead to gyroscope saturation (are they?).

@ParthaEth: I never observed that Rovio turns in the wrong direction with slow rotations. This seems weird to me.

closed 2019-10-07 by xiaopenghuang