ethz-asl/rovio #128: Help With Bosch BN0055 IMU

patrickHow, 2017-08-30:
Hi All,

I'm trying to get Rovio running with a Bosch BNO055 IMU on a Tinkerforge 2.0 Brix board. I can't seem to find random walk data anywhere on the datasheet, but I think I've read people have used this sensor in the past to run Rovio. Any help on finding the required values?

andre-nguyen, 2017-08-30:
Here are some ressources that can help you out

* https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model
* https://www.vectornav.com/support/library/gyroscope

Pay attention to the units so you can convert them depending on how the manufacturer lists the data. You can build the allan variance plot yourself or use some existing tools to do it e.g. https://github.com/GAVLab/allan_variance (never tested this tool myself though)

closed 2018-01-11 by bloesch

raoulj, 2018-05-02:
As an alternative to GAVLab's Allan variance tool, you can also use [this tool](https://github.com/rpng/kalibr_allan), which determines the Allan Variance and regresses appropriately for you.