ethz-asl/rovio #129: Derivations of Equation 9 and Equation 10 in the paper

BITVoyager, 2017-08-31:
Hi. Could you please provide any insights on the derivations of Equation 9 and Equation 10 in the paper? Thank you.

bloesch, 2017-09-06:
There will be some more details in a journal paper that should get accessible this month. I will link it as soon as its available online.

BITVoyager, 2017-09-06:
Cool. Thank you. Do you have a short answer to this question? :-)

bloesch, 2017-09-06:
I do not think there is a short answer to this. What I did is introduce a parametrization for unit vectors together with a representation of differences between them. The differentials then can be derived by looking at the limit of small perturbations on those unit vectors (similar to the classical derivation of differentials)...

bloesch, 2017-09-19:
I link to the IJRR paper is provided in the README. Hope that helps.

BITVoyager, 2017-09-20:
@bloesch Thank you.

closed 2017-09-20 by BITVoyager