ethz-asl/rovio #131: SrvResetToPose.h missing

instantbling, 2017-09-12:
Hi,

I was trying to install your packages, but rovio kept giving me this error  `fatal error: rovio/SrvResetToPose.h: No such file or directory`. 
I have tried what you said previously, building after the first failure, but there was no difference. The other packages built just fine.

I am unable to perform on a fresh install of ubuntu as there are already many other things installed inside. Could you perhaps provide a copy of the file and a list of the items inside that should be changed to match the pc? Or is there any other methods I might be able to try?

Thank you very much!

marco-tranzatto, 2017-09-12:
Hi,
The file is already cinluded in the repo: [srv](https://github.com/ethz-asl/rovio/blob/master/srv/SrvResetToPose.srv)

Which Ubuntu are you using?

instantbling, 2017-09-13:
Hi, thank you very much for the quick response :)
I'm using Ubuntu 14.04.5 LTS
Can I just rename the file extension and place it in the correct folder?

marco-tranzatto, 2017-09-13:
Hi,
it sounds strange that you can't compile rovio at all. Are you using `catkin build` command?

Also, the file is already in the "right" folder, as we are able to compile and use rovio without any issue.
Nevertheless you can do whatever you want in your fork of the repo ;)

instantbling, 2017-09-13:
Hi,
Sorry it took so long. I tried that and got a whole set of new errors. :"(
I'm thinking the file might be incomplete. 
Yes, I used the `catkin build` command.
I have attached the errors and my SrvResetToPose.h (renamed to .txt as the site doesn't accept .h files) here for your reference. Are the contents of SrvResetToPose correct? Or am I doing something wrongly? 

[rovioerror.txt](https://github.com/ethz-asl/rovio/files/1299302/rovioerror.txt)
[SrvResetToPose.txt](https://github.com/ethz-asl/rovio/files/1299310/SrvResetToPose.txt)

marco-tranzatto, 2017-09-14:
I doubt the file is incomplete. We can compile rovio under Ubuntu 14.04 and use this service.
Some questions:
- Are you compiling the latest version of master?
- Are you following these [instruction](https://github.com/ethz-asl/rovio#install-without-opengl-scene)?
- Did you modify anything?

Service definition that you attached is exactly the same as the one in the repository ...

instantbling, 2017-09-14:
Thanks for the quick response :)
I used the instructions [here](https://github.com/ethz-asl/mav_dji_ros_interface/wiki/Ubuntu-and-ROS). All instructions were copied and pasted so I shouldn't have missed on anything...installing kindr was a pain though ^_^"
No I haven't modified anything :(

I'm going to try building it by itself in a separate folder instead as per the instructions you posted and keep you updated.

instantbling, 2017-09-14:
^_^"" seems the instruction worked!
I tried building both separately and within the dji ros interface project folder, seems to have turned out fine, 2 blacklisted packages and 47 packages successfully built :D
Seems the instructions on the ros dji interface site may be a little outdated.

Thanks a lot for all the help!

closed 2017-09-14 by instantbling