ethz-asl/rovio #132: predictMergedEKF mean measurement error

stihuangyuan, 2017-09-13:
thanks for your great work. When i look into imu predict,  calculate mean measurent
`double t = itMeasStart->first;
        for(typename std::map<double,mtMeas>::const_iterator itMeas=next(itMeasStart); itMeas!=itMeasEnd; itMeas++)
        {
            itMeasStart->second.boxMinus(itMeas->second,difVec);
            vec = vec + difVec*(std::min(itMeas->first,tTarget)-t);
            t = std::min(itMeas->first,tTarget);
        }
        vec = vec/dT;
        itMeasStart->second.boxPlus(vec,meanMeas);`
is there a symbol error? maybe it should be this?
`itMeasStart->second.boxMinus(vec,meanMeas);`

bloesch, 2017-09-13:
It looks like there is a sign error there. The third line should do itMeas - itMeasStart and not the other way round. I will check this and create a PR. Thanks for the catch.

bloesch, 2017-09-13:
see #133. Thanks again!

closed 2017-09-19 by bloesch