ethz-asl/rovio #134: Why do ekf update every visible feature, not do one time for all visible features

xiaopenghuang, 2017-09-18:
In paper, I see you said stack all the error term then do EKF update, but in code you do it everytime a single visible feature. Which is consume more time , and if the one feature is not right detect, it can affect the whole system which is rather bad.

bloesch, 2017-09-19:
Originally all feature were updated at the same time. I agree that if the first couple of landmarks are badly tracked it might lead to failure. On the other hand if the first features are well tracked this will help tracking of the subsequent features and decrease their initial uncertainty (which also dictates computation time). Furthermore, if I remember right, doing N times 2-dimensional updates, is faster than doing one (N*2)-dimensional update due to the inversion of the innovation matrix.

xiaopenghuang, 2017-09-19:
>Furthermore, if I remember right, doing N times 2-dimensional updates, is faster than doing one (N*2)-dimensional update due to the inversion of the innovation matrix.
The above 2-dimensional updates you mean in calculate the update feature corrdinates(SVD decomposion ), but in ekf update ,you also need to calculate the matrix inverse.
Is it really faster to do ekf update every feature.

bloesch, 2017-09-19:
The matrix inversion is faster in the case where you split the measurements. In the end however, the speed strongly depends on how much of the sparsity can be exploited in either case. So you might be able to go with a full update.

closed 2017-09-19 by xiaopenghuang