ethz-asl/rovio #135: Segfault on Ubuntu 16.04

tomlankhorst, 2017-11-09:
I'm trying to send Optitrack pose data to Rovio to help it align its coordinate frame. The Optitrack data is PoseStamped so I added a subscriber that allows for PoseStamped instead of TransformStamped (originally Rovio's /pose subscriber). 

For details, check [this diff. ](https://gist.github.com/tomlankhorst/3fdd66087601be041c1f9ac37b12ad3b)

This however does not seem to be core to the following segfault I'm experiencing...

I got Rovio working without the pose input. With pose input, however, the process stops with signal 11 (SIGSEGV). I recompiled Rovio and ran it with GDB. This resulted in the following run with stacktrace.

```
[ WARN] [1510241808.320064726, 1505128145.535715277]: ====================== Debug Mode ======================
Testing Prediction
==== Model jacInput Test failed: 1 is larger than 1e-06 at row 15(vep) and col 15(vep) ====
  1  0
==== Test successful (1.35254e-07) ====
Testing cameraOutputCF
==== Test successful (3.57173e-08) ====
Testing imuOutputCF
==== Test successful (1.27552e-08) ====
Testing attitudeToYprCF
==== Test successful (6.43838e-08) ====
Testing transformFeatureOutputCT
==== Test successful (1.41033e-08) ====
==== Test successful (6.221e-08) ====
Testing LandmarkOutputImuCT
==== Test successful (2.8556e-08) ====
Testing FeatureOutputReadableCT
==== Test successful (9.12083e-09) ====
Testing pixelOutputCT (can sometimes exhibit large absolut errors due to the float precision)
==== Model jacInput Test failed: 5.98709 is larger than 1 at row 0(def_0) and col 1() ====
  24161.3  24155.3
Testing zero velocity update
==== Test successful (1.39778e-10) ====
==== Test successful (1.39778e-10) ====
-- Filter: Initializing using accel. measurement ...
-- Filter: Initialized at t = 1505128145.57

Thread 1 "rovio_node" received signal SIGSEGV, Segmentation fault.
0x0000000001046506 in _mm256_load_pd (__P=0x3b6ba90) at /usr/lib/gcc/x86_64-linux-gnu/5/include/avxintrin.h:836
836       return *(__m256d *)__P;
(gdb) bt
#0  0x0000000001046506 in _mm256_load_pd (__P=0x3b6ba90) at /usr/lib/gcc/x86_64-linux-gnu/5/include/avxintrin.h:836
#1  Eigen::internal::pload<double __vector(4)>(Eigen::internal::unpacket_traits<double __vector(4)>::type const*) (from=0x3b6ba90) at /usr/include/eigen3/Eigen/src/Core/arch/AVX/PacketMath.h:207
#2  0x000000000115fb28 in Eigen::internal::ploadt<double __vector(4), 32>(Eigen::internal::unpacket_traits<double __vector(4)>::type const*) (from=0x3b6ba90) at /usr/include/eigen3/Eigen/src/Core/GenericPacketMath.h:475
#3  Eigen::internal::evaluator<Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> > >::packet<32, double __vector(4)>(long, long) const (this=0x7fffffff5e08, row=0, col=0) at /usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h:183
#4  0x000000000114d86d in Eigen::internal::unary_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, 4, 1, 0, 4, 1> const>, Eigen::internal::IndexBased, double>::packet<32, double __vector(4)>(long, long) const (this=0x7fffffff5e00, row=0, col=0)
    at /usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h:406
#5  0x0000000001131f64 in Eigen::internal::redux_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, 4, 1, 0, 4, 1> const> >::packetByOuterInner<32, double __vector(4)>(long, long) const (this=0x7fffffff5e00, outer=0, inner=0) at /usr/include/eigen3/Eigen/src/Core/Redux.h:383
#6  0x000000000111bd59 in Eigen::internal::redux_vec_unroller<Eigen::internal::scalar_sum_op<double>, Eigen::internal::redux_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, 4, 1, 0, 4, 1> const> >, 0, 1>::run (mat=...) at /usr/include/eigen3/Eigen/src/Core/Redux.h:170
#7  0x0000000001107902 in Eigen::internal::redux_impl<Eigen::internal::scalar_sum_op<double>, Eigen::internal::redux_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, 4, 1, 0, 4, 1> const> >, 3, 2>::run (mat=..., func=...) at /usr/include/eigen3/Eigen/src/Core/Redux.h:319
#8  0x00000000010f2ab0 in Eigen::DenseBase<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, 4, 1, 0, 4, 1> const> >::redux<Eigen::internal::scalar_sum_op<double> > (this=0x7fffffff5ea0, func=...) at /usr/include/eigen3/Eigen/src/Core/Redux.h:416
#9  0x00000000010dec33 in Eigen::DenseBase<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs2_op<double>, Eigen::Matrix<double, 4, 1, 0, 4, 1> const> >::sum (this=0x7fffffff5ea0) at /usr/include/eigen3/Eigen/src/Core/Redux.h:449
#10 0x00000000010c87c4 in Eigen::MatrixBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> >::squaredNorm (this=0x3b6ba90) at /usr/include/eigen3/Eigen/src/Core/Dot.h:90
#11 0x00000000010b0e1e in Eigen::MatrixBase<Eigen::Matrix<double, 4, 1, 0, 4, 1> >::norm (this=0x3b6ba90) at /usr/include/eigen3/Eigen/src/Core/Dot.h:103
#12 0x00000000010b0bda in Eigen::QuaternionBase<Eigen::Quaternion<double, 0> >::norm (this=0x3b6ba90) at /usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:120
#13 0x0000000001093e18 in kindr::Quaternion<double>::norm (this=0x3b6ba90) at /home/aeroworks/aeroworks_ws/devel/include/kindr/quaternions/Quaternion.hpp:223
#14 0x000000000107811c in kindr::UnitQuaternion<double>::norm (this=0x3b6ba90) at /home/aeroworks/aeroworks_ws/devel/include/kindr/quaternions/Quaternion.hpp:474
#15 0x0000000001066fbe in kindr::UnitQuaternion<double>::UnitQuaternion (this=0x3b6ba90, other=...) at /home/aeroworks/aeroworks_ws/devel/include/kindr/quaternions/Quaternion.hpp:353
#16 0x0000000001053cb9 in kindr::RotationQuaternion<double>::RotationQuaternion (this=0x3b6ba90) at /home/aeroworks/aeroworks_ws/devel/include/kindr/rotations/RotationQuaternion.hpp:84
#17 0x000000000104ac29 in LWF::QuaternionElement::QuaternionElement (this=0x3b6ba50) at /home/aeroworks/aeroworks_ws/src/rovio/lightweight_filtering/include/lightweight_filtering/State.hpp:186
#18 0x00000000010570e2 in std::_Head_base<1ul, LWF::QuaternionElement, false>::_Head_base (this=0x3b6ba50) at /usr/include/c++/5/tuple:105
#19 0x000000000105711a in std::_Tuple_impl<1ul, LWF::QuaternionElement>::_Tuple_impl (this=0x3b6ba50) at /usr/include/c++/5/tuple:353
#20 0x0000000001057153 in std::_Tuple_impl<0ul, LWF::VectorElement<3u>, LWF::QuaternionElement>::_Tuple_impl (this=0x3b6ba50) at /usr/include/c++/5/tuple:202
#21 0x00000000010571ca in std::tuple<LWF::VectorElement<3u>, LWF::QuaternionElement>::tuple (this=0x3b6ba50) at /usr/include/c++/5/tuple:602
#22 0x0000000001057213 in LWF::State<LWF::VectorElement<3u>, LWF::QuaternionElement>::State (this=0x3b6ba30) at /home/aeroworks/aeroworks_ws/src/rovio/lightweight_filtering/include/lightweight_filtering/State.hpp:514
#23 0x000000000104e9b7 in rovio::PoseUpdateMeas::PoseUpdateMeas (this=0x3b6ba30) at /home/aeroworks/aeroworks_ws/src/rovio/include/rovio/PoseUpdate.hpp:64
#24 0x000000000115a152 in std::pair<double const, rovio::PoseUpdateMeas>::pair<double const&, 0ul>(std::tuple<double const&>&, std::tuple<>&, std::_Index_tuple<0ul>, std::_Index_tuple<>) (this=0x3b6ba10, __tuple1=std::tuple containing = {...}, __tuple2=empty std::tuple) at /usr/include/c++/5/tuple:1172
#25 0x0000000001147264 in std::pair<double const, rovio::PoseUpdateMeas>::pair<double const&>(std::piecewise_construct_t, std::tuple<double const&>, std::tuple<>) (this=0x3b6ba10, __first=std::tuple containing = {...}, __second=empty std::tuple) at /usr/include/c++/5/tuple:1161
#26 0x000000000112a357 in __gnu_cxx::new_allocator<std::_Rb_tree_node<std::pair<double const, rovio::PoseUpdateMeas> > >::construct<std::pair<double const, rovio::PoseUpdateMeas>, std::piecewise_construct_t const&, std::tuple<double const&>, std::tuple<> >(std::pair<double const, rovio::PoseUpdateMeas>*, std::piecewise_construct_t const&, std::tuple<double const&>&&, std::tuple<>&&) (this=0x7fffcab42248, __p=0x3b6ba10) at /usr/include/c++/5/ext/new_allocator.h:120
#27 0x0000000001112abc in std::allocator_traits<std::allocator<std::_Rb_tree_node<std::pair<double const, rovio::PoseUpdateMeas> > > >::construct<std::pair<double const, rovio::PoseUpdateMeas>, std::piecewise_construct_t const&, std::tuple<double const&>, std::tuple<> >(std::allocator<std::_Rb_tree_node<std::pair<double const, rovio::PoseUpdateMeas> > >&, std::pair<double const, rovio::PoseUpdateMeas>*, std::piecewise_construct_t const&, std::tuple<double const&>&&, std::tuple<>&&) (__a=..., __p=0x3b6ba10) at /usr/include/c++/5/bits/alloc_traits.h:530
[... cut ...]
This seems to be some kind of alignment problem that occurs when trying to vectorize matrix math. I tried to add `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` ([to help Eigen align members of classes](https://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html)) to the remaining classes defined in `RovioNode.hpp` but this did not seem to help. 

Did anyone experience this problem and if so, were you able to fix it? 

Seems to be related to: 

https://github.com/ethz-asl/rovio/issues/94
https://github.com/ethz-asl/rovio/issues/103
https://github.com/ethz-asl/rovio/pull/119

tomlankhorst, 2017-11-22:
As mentioned in #83 , disabling Eigen's static alignment `-DEIGEN_DONT_ALIGN_STATICALLY` solves the problem but negatively impacts performance. 

So, as suspected, this seems to be related to alignment issues.

bloesch, 2017-11-23:
We thought we fixed alignment related issues in #119. But it does seem to remain alignment related, unfortunately adding EIGEN_MAKE_ALIGNED_OPERATOR_NEW did not help much in the past. There might be another missing stl allocator (like in #119)...

tomlankhorst, 2017-11-23:
@bloesch Thank you for your response. It seems that the branch `fix/MalfunctioningPoseUpdate` solves it for me. I merged master back into `fix/MalfunctioningPoseUpdate` and needed to add `MaxFeatureFrames` (10000 default value) in `ImgUpdate`. 

Is there a reason why we don't merge this branch? :)

Runepx4, 2020-12-17:
I am getting the same error (in ubuntu 18.04) when using a ground pose. @tomlankhorst, did you manage to solve the error back in 2017?