ethz-asl/rovio #137: n.getM() in bearingToPixel

liucsg, 2017-11-29:
Hi bloesch!
As I understand, Camera::bearingToPixel is doing projection(3D -> 2D), I do not know what is mean of n.getM(). Could you help giving some guidance on this item?
Thanks

bloesch, 2017-12-01:
getM returns two perpendicular directions to the bearing vector n. Relating to http://dx.doi.org/10.1177/0278364917728574 , it represents a rotated version of the matrix N. It is used to parametrize local perturbations on the bearing vector n.

liucsg, 2017-12-01:
bool Camera::bearingToPixel(const LWF::NormalVectorElement& n, cv::Point2f& c, Eigen::Matrix<double,2,2>& J) const{
Eigen::Matrix<double,3,2> J1;
J1 = n.getM();
Eigen::Matrix<double,2,3> J2;
const bool success = bearingToPixel(n.getVec(),c,J2);
J = J2*J1;
return success;
}

Hi bloesch,
I know the mean of J2. Could you tell me that why J is equal to J2*J1 In this function?
Thanks!

bloesch, 2017-12-04:
This is the chain rule:

J2: d(pixel)/d(global coordinates)
J1: d(global coordinates)/d(local coordinates)
=> J = d(pixel)/d(local coordinates)

If you are not familiar with the concept of local/global coordinates the following may help:
- http://asrl.utias.utoronto.ca/~tdb/bib/barfoot_ser17.pdf
- http://ceres-solver.org/nnls_modeling.html#localparameterization
- https://arxiv.org/abs/1606.05285

liucsg, 2017-12-04:
Thank you very much!

http://asrl.utias.utoronto.ca/~tdb/bib/barfoot_ser17.pdf
http://ceres-solver.org/nnls_modeling.html#localparameterization
https://arxiv.org/abs/1606.05285

closed 2018-01-11 by bloesch