ethz-asl/rovio #138: Rovio and more than 2 cameras

Cezary-Ziemiecki, 2017-12-05:
Hi,
Does anybody know anything about working Rovio with 3 or more cameras? It could be very helpfull to share every small information about it.
However exactly I mean about set of 3 cameras at very various angles. Main camera is looking on the front of robot and two of them are looking on the right and left side of robot. It's difficult to use kalibr for calibration them but apart of it, Do You know if it is possible to use such set of cameras?

Thanks
Cezary

bloesch, 2017-12-05:
Never tested, but it should be supported. You will have to add Camera2 to rovio.info. Concerning extrinsics, they can be estimated online to some extent. But if you get a good initial guess you will increase the chance that it works.

bloesch, 2017-12-05:
Also, all cameras and the IMU have to be time synchonized, i.e., the timestamps have to be expressed w.r.t. a common clock!

raoulj, 2018-05-01:
@Cezary-Ziemiecki Was this successful?  I'm attempting something similar right now.