ethz-asl/rovio #140: Obvious drift on Z direction comparing with tango

BryceeeeeChen, 2017-12-22:
Hello, when I compared the performance of rovio with tango, I found that the odometry on Z direction would drift obviously. I have already checked an old issue that might specified the similar problem(#23 ) but it did not help. 
The problem can be seen in figures below. The blue line was generated by rovio using realsense ZR300, the red line was generated by tango (Lenovo Phab 2 Pro) and both were running without loop closure/relocalization. I was holding these two devices bonded together and walked along a 6mx4.2m rectangle 4 times then returned the start point. 

[image]
 
[image]

Could someone help me solve this problem? Thanks...

bloesch, 2018-01-04:
There will always be some drift, so the questions is how to keep it minimal. The main answer is that you will have to tune the rovio parameters to the system (camera/imu) you are using. If I remember right, we could improve performance on the ZR300 by increasing the accelerometer covariance parameter: try setting vel_* to something like 4e-5 or 4e-4?

BryceeeeeChen, 2018-01-09:
Sorry for my late reply. I tuned the covariance parameter(vel_0/1/2) as you said, but the result was same. The odometry still drifted up "regularly" when the camera was moving, did you guys see the same behavior while using ZR300? I understand that drift is inevitable for an open-loop odometry system, but what is the most possible cause for regularly drifting up? Should I try to tune all parameters related to accelerometer bias of Z? Thank you.

bloesch, 2018-01-11:
This is quite difficult to answer. This does not only look like random drift but more like a constant bias upwards. Is it independent of the trajectory you have? Do you have other sensor, and if yes, did you observe a similar behavior?

It has been a while that we tested rovio with the ZR300 and I remember that back then we had driver related issues which were causing some of the IMU outputs to be set to zero from time to time. Maybe if you could just quickly output and doublecheck the IMU measurements you are sending to rovio? Also, in steady-state, is the norm of the accelerometer approximately 9.81?

BryceeeeeChen, 2018-01-15:
@bloesch Thank you. I did this experiment many times (walking along the rectangle pattern in counterclockwise direction). All trajectories constantly drift up. But today I found an interesting thing, when I walked in clockwise direction, the trajectory would constantly drift down. Do you have any clue on this behavior?   
[image]
I haven't tried other sensor, but I have two ZR300 cameras. Both gave me same results.

The IMU gave me continuous values. But the gravity norm I got from ZR300 filtered by very aggressive alpha beta low pass was around 9.7 most of the time (I am in Hong Kong, the local gravity value is around 9.785).....would it caused the drift problem? If I largely pitched the camera down, the measurement value could reach 10. I believe it is because the accelerometer in ZR300 has different biases in different orientation. Would it also be a crucial issue? And Could ROVIO estimate biases on the fly?

NikolausDemmel, 2018-03-14:
>  All trajectories constantly drift up. But today I found an interesting thing, when I walked in clockwise direction, the trajectory would constantly drift down. Do you have any clue on this behavior?

This sounds like it could be a rolling shutter effect to me. Are you using the color camera of the ZR300 instead of the fisheye one?

To test this further, you could also try rotating the camera by 90 degrees and see if the systematic translation drift also rotates (i.e. if it is inverted if you hold the camera upside down).

BryceeeeeChen, 2018-03-16:
Hi @NikolausDemmel, the fisheye was what I used back then. I will try holding the camera with 90 degrees rotation, it seems like a smart way to test which part of my original calibration parameters were slightly off. I forgot to point out that I never encountered this problem after using the calibration parameters provided by ASL with rovioli in pure VIO mode (so it is rovio). But the sudden acceleration issue is still there sometimes, much less chance though (what we discussed in the maplab thread). Therefore, I might close this issue now because of bad calibration. Thank you.

closed 2018-03-16 by BryceeeeeChen

NikolausDemmel, 2018-03-18:
@BryceeeeeChen, if you use the fisheye, that is global shutter. But you are right, one can imaging that inaccurate (extrinsic) calibration could also cause such a systematic drift.