ethz-asl/rovio #141: catkin build rovio failed

zhouchang1993, 2017-12-25:
Hi I build rovio failed. 
Here's the wrong message.
CMakeFiles/rovio_node.dir/src/rovio_node.cpp.o: In function `void LWF::OutlierDetectionBase<0u, 2u>::check<2>(Eigen::Matrix<double, 2, 1, ((Eigen::._104)0)|((((2)==(1))&&((1)!=(1)))?((Eigen::._104)1) : ((((1)==(1))&&((2)!=(1)))?((Eigen::._104)0) : ((Eigen::._104)0))), 2, 1> const&, Eigen::Matrix<double, -1, -1, 0, -1, -1> const&)':
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0xff): undefined reference to `google::kLogSiteUninitialized'
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x11c): undefined reference to `google::LogMessage::LogMessage(char const*, int)'
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x126): undefined reference to `google::LogMessage::stream()'
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x189): undefined reference to `google::LogMessage::~LogMessage()'
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x1bb): undefined reference to `fLI::FLAGS_v'
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x1c5): undefined reference to `google::InitVLOG3__(int**, int*, char const*, int)'
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj2EE5checkILi2EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x1f6): undefined reference to `google::LogMessage::~LogMessage()'
CMakeFiles/rovio_node.dir/src/rovio_node.cpp.o: In function `void LWF::OutlierDetectionBase<0u, 3u>::check<3>(Eigen::Matrix<double, 3, 1, ((Eigen::._104)0)|((((3)==(1))&&((1)!=(1)))?((Eigen::._104)1) : ((((1)==(1))&&((3)!=(1)))?((Eigen::._104)0) : ((Eigen::._104)0))), 3, 1> const&, Eigen::Matrix<double, -1, -1, 0, -1, -1> const&)':
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x170): undefined reference to `google::kLogSiteUninitialized'
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x193): undefined reference to `google::LogMessage::LogMessage(char const*, int)'
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x19d): undefined reference to `google::LogMessage::stream()'
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x200): undefined reference to `google::LogMessage::~LogMessage()'
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x22b): undefined reference to `fLI::FLAGS_v'
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x235): undefined reference to `google::InitVLOG3__(int**, int*, char const*, int)'
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj3EE5checkILi3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x266): undefined reference to `google::LogMessage::~LogMessage()'
CMakeFiles/rovio_node.dir/src/rovio_node.cpp.o: In function `void LWF::OutlierDetectionBase<0u, 6u>::check<6>(Eigen::Matrix<double, 6, 1, ((Eigen::._104)0)|((((6)==(1))&&((1)!=(1)))?((Eigen::._104)1) : ((((1)==(1))&&((6)!=(1)))?((Eigen::._104)0) : ((Eigen::._104)0))), 6, 1> const&, Eigen::Matrix<double, -1, -1, 0, -1, -1> const&)':
rovio_node.cpp:(.text._ZN3LWF20OutlierDetectionBaseILj0ELj6EE5checkILi6EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE[_ZN3LWF20OutlierDetectionBaseILj0ELj6EE5checkILi6EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._104E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEE]+0x76b): undefined reference to `google:: [... cut ...] 3EEEvRKN5Eigen6MatrixIdXT_ELi1EXorLNS3_5._179E0EquaaeqT_Li1EneLi1ELi1ELS5_1EquaaeqLi1ELi1EneT_Li1ELS5_0ELS5_0EEXT_ELi1EEERKNS4_IdLin1ELin1ELi0ELin1ELin1EEEE8vlocal__]+0x0): undefined reference to `google::kLogSiteUninitialized'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/indemind/catkin_ws/devel/lib/rovio/rovio_rosbag_loader] Error 1
make[1]: *** [CMakeFiles/rovio_rosbag_loader.dir/all] Error 2
make: *** [all] Error 2
glog has been installed
I hope to give me some help ,thanks

bloesch, 2018-01-04:
not sure what this is, where does the glog linking come from?

plumewind, 2018-09-14:
Hello, is it solved? I also encountered this problem.