ethz-asl/rovio #143: roslaunch error

suzy419, 2018-01-02:
I run this package in Ubuntu 14.04 indigo,when use "roslaunch rovio rovio_rosbag_node.launch ",it shows some problems.

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /rovio/cam0_topic_name: /cam0/image_raw
 * /rovio/cam1_topic_name: /cam1/image_raw
 * /rovio/camera0_config: /home/suzy/catkin...
 * /rovio/camera1_config: /home/suzy/catkin...
 * /rovio/filter_config: /home/suzy/catkin...
 * /rovio/imu_topic_name: /imu0
 * /rovio/rosbag_filename: /home/suzy/datase...

NODES
  /
    rovio (rovio/rovio_rosbag_loader)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rovio-1]: started with pid [13211]
[rovio-1] process has died [pid 13211, exit code -11, cmd /home/suzy/catkin_ws/devel/lib/rovio/rovio_rosbag_loader __name:=rovio __log:=/home/suzy/.ros/log/af4e6b98-efbb-11e7-a5ff-54ee75d58232/rovio-1.log].
log file: /home/suzy/.ros/log/af4e6b98-efbb-11e7-a5ff-54ee75d58232/rovio-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

closed 2018-01-02 by suzy419

AjSat, 2018-03-23:
@suzy419 

Hello, I am obtaining the same error when I use "roslaunch rovio rovio_rosbag_node.launch ". Could you please tell me how you resolved it?

Thank you in advance

suzy419, 2018-03-24:
Reinstall the Ubuntu system 307045795 邮箱：307045795@qq.com 签名由 网易邮箱大师 定制 在2018年03月24日 03:42，Ajay Sathya 写道： @suzy419 Hello, I am obtaining the same error when I use "roslaunch rovio rovio_rosbag_node.launch ". Could you please tell me how you resolved it? Thank you in advance