ethz-asl/rovio #144: ROVIOLI in MAPLAB

zz1120122573, 2018-01-04:
I want to run ROVIOLI in stereo mode, but maplab owner said "ROVIO has some issues in stereo mode, we don't recommend using it if you don't need it". Are the problems important? Can I solve them?

bloesch, 2018-01-04:
Not too sure what the issues are. The stereo mode is relatively sensitive to bad calibrations and often there is no significant gain in performance when compared to the monocular setup. I am not 100% sure whether the calibration sensitivity is the only cause for problems though...

zz1120122573, 2018-01-09:
Thanks for your reply. Do other stereo VIO sensitive to bad calibrations? e.g. OKVIS. OKVIS will lost in the middle when running Euroc dataset V2_03.

bloesch, 2018-01-11:
Many VIO system I know are sensitive to bad calibration. It depends a bit on the method and on whether on-line calibration is supported. But often, even with online calibration, the system are not performing well if the calibration is bad.