ethz-asl/rovio #146: How to get the dataset trajectory the method generated

cumtchenchang, 2018-01-11:
I want to evelate the method with our lab method. How to change the code to get dataset trajectory the method generated.

bloesch, 2018-01-11:
Please have a look at the rostopic output.

cumtchenchang, 2018-01-21:
@bloesch  Thank you for your help. I change the code folled what you said, but It dose not compile.

Kushine, 2018-01-24:
How to get dataset trajectory the method generated?Have you got it? @cumtchenchang

cumtchenchang, 2018-01-24:
@Kushine I got it. But I am worry it differs from the authors.

Kushine, 2018-01-26:
In RovioNode.hpp, you can get dataset trajectory from " imuOutput_ ".

cumtchenchang, 2018-01-26:
@Kushine  can you give me your email? my is 1253419722@qq.com

cumtchenchang, 2018-01-26:
@Kushine  I can not change the code correctly to get the trajectory.

Kushine, 2018-01-30:
In RovioNode.hpp, save imuOutput_ to a vector

 in RovioNode.hpp,
after line 670 imuOutputCT_.transformState(state,imuOutput_); 
add the code 
std::cout <<std::fixed<<mpFilter_->safe_.t_<< "," << imuOutput_.WrWB()(0) << "," << imuOutput_.WrWB()(1)<< "," << imuOutput_.WrWB()(2)<< std::endl;
        std::cout <<std::fixed<<imuOutput_.qBW().x()<< "," <<imuOutput_.qBW().y()<< "," <<imuOutput_.qBW().z()<< "," <<-imuOutput_.qBW().w()<< std::endl;

Kushine, 2018-01-30:
In RovioNode.hpp, save imuOutput_ to a vector, or print as the way: 
in RovioNode.hpp,
after line 670 imuOutputCT_.transformState(state,imuOutput_);
add the code
std::cout <<std::fixed<<mpFilter_->safe_.t_<< "," << imuOutput_.WrWB()(0) << "," << imuOutput_.WrWB()(1)<< "," << imuOutput_.WrWB()(2)<< std::endl;
std::cout <<std::fixed<<imuOutput_.qBW().x()<< "," <<imuOutput_.qBW().y()<< "," <<imuOutput_.qBW().z()<< "," <<-imuOutput_.qBW().w()<< std::endl;

cumtchenchang, 2018-01-30:
@Kushine can you give more detail? I cannot save the trajectory to txt or other files. thank you

jixiedaima, 2018-04-11:
@Kushine  Can you tell me more?

jixiedaima, 2018-05-07:
@Kushine  solved it, thanks!

habib-bolourchi, 2018-08-07:
@Kushine would you tell me if imuOutput_ contains the vio answer or just the numbers that come from IMU (the device)?

Kushine, 2018-08-08:
imuOutput_ contains the vio answer 

顾升宇   算法工程师
视辰信息科技（上海）有限公司
www.sightp.com
Email：gushengyu@sightp.com
Tel:18862918665
Add:上海市浦东新区张江镇郭守敬路498号23号楼5层
 
From: habib boloorchi
Date: 2018-08-08 03:19
To: ethz-asl/rovio
CC: Kushine; Mention
Subject: Re: [ethz-asl/rovio] How to get the dataset trajectory the method generated (#146)
@Kushine would you tell me if imuOutput_ contains the vio answer or just the numbers that come from IMU (the device)?

closed 2019-02-17 by bloesch

better-fcl, 2020-04-22:
> In RovioNode.hpp, save imuOutput_ to a vector, or print as the way:
> in RovioNode.hpp,
> after line 670 imuOutputCT_.transformState(state,imuOutput_);
> add the code
> std::cout [...]

thank you very much!!!!!!

better-fcl, 2020-04-22:
> In RovioNode.hpp, save imuOutput_ to a vector, or print as the way:
> in RovioNode.hpp,
> after line 670 imuOutputCT_.transformState(state,imuOutput_);
> add the code
> std::cout [...]

But i still want to know how to set the route of the result (pose.txt), can you help me? Plz, thank you!