ethz-asl/rovio #151: How to run this package ?

SrijaneeBiswas, 2018-02-09:
Hi All,

I'm using Ubuntu 16.04 and ROS Kinetic system, and I have git cloned the package into my workspace. But I have found no documentation regarding what nodes to run, what topics to publish and so on. Basically, I don't know how I should proceed after having installed the package. Any guidance on this would be helpful.

Thanks,
Srijanee

marco-tranzatto, 2018-02-10:
Hi,

I suggest you start downloading the Euroc dataset here https://github.com/ethz-asl/rovio#euroc-datasets. The you can play any rosbag in that data set by using [rosbag play](http://wiki.ros.org/rosbag/Commandline#play) and finally in a new terminal you can launch the rovio node:

`roslaunch rovio rovio_node.launch`

Regards,
Marco.

FPSychotic, 2018-02-11:
Same problem, I want to run rovio in a arm based robot and see the data in realtime, not a bag with historic data. There is not launch file in rovio folders,nothing how setting the camera about,and so. Im totally lost, somebody must to do something to document better the use of the software and how install it in arm,its crazy really, with all respect of the world,is needed be a genius of linux, cmake,catkin to get something with the ethz software, im totally frustated,i cant install mablap,svo 2.0,etc, just rovio, and I cant run it

marco-tranzatto, 2018-02-11:
Hey, we do our best to put online usable code and are always keen to receive positive criticism and help people with their issues. Said that, if you demand for something or say "somebody must do this and that" what do you think you'll get in return?

If something is not clear, or not enough documented, then ask for clarifications, in a polite way. Then you could even open a pull request to update the documentation and contribute directly and improve the repositories.
The software here is the outcome of many people working very hard is FREELY available to everyone ... take a moment to appreciate this ;)

SrijaneeBiswas, 2018-02-11:
Hi Marco,

Thanks for being prompt with your response. I mentioned in my last comment that I have installed the package but I realized later that unfortunately, I couldn't build the package. 'Catkin_make' fails (probably due to a versioning issue?). Can you guide me on how to install the package for an Ubuntu 16.04, ROS Kinetic system? 

It'd also be great if you can tell me how to run this without a bag file. Say, I have my own feature detection ROS node that publishes the pixel locations of these features. I want to pass this through the rovio tracker so that they could be tracked real-time.

Thanks,
Srijanee

marco-tranzatto, 2018-02-11:
Hi Srijanee,

I suggest you use [catkin build](http://catkin-tools.readthedocs.io/en/latest/verbs/catkin_build.html) instead of catkin_make.

For your second question: ROVIO extracts features from the image topic "cam0/image_raw" (see [here](https://github.com/ethz-asl/rovio/blob/master/include/rovio/RovioNode.hpp#L209)), so if you want to use your own features you'd need to modify ROVIO to do that. Why don't you let ROVIO extract the features for you?

Regards,
Marco.

SrijaneeBiswas, 2018-02-11:
Hi Marco,

Sorry, catkin build seems to fail as well (I've attached a screenshot for you to see)
[image]

I'm new to catkin build tools as I've only used catkin_make to build my packages. I'll be glad if you can help me through this.

Regarding using ROVIO, I can do that. I assume the only thing that'll change is the image topic name (instead of cam0/image_raw). Also, does the package handle data association of these features as well?

Regards,
Srijanee

ZacharyTaylor, 2018-02-11:
Hi Srijanee,
The issue here is that rovio depends on lightweight filtering which is a submodule of the git repo. You need to ensure it is also checked out. cd to the rovio directory and run `git submodule update --init --recursive`

ZacharyTaylor, 2018-02-11:
Also rovio will take in an stream of images and imu messages and output the estimated odometry. All feature association and such is done internally. However, to run it on your own data you should probably be aware that rovio needs quite accurate time-stamping, some calibration information and a high quality IMU to run reliably.

SrijaneeBiswas, 2018-02-12:
Hi Zachary,

cmake fails at the rovio package even after running the git submodule command. Attached is a screenshot of the same
[image]

ZacharyTaylor, 2018-02-12:
Hi  Srijanee
Lightweight filtering if failing on link 12 of its CMakeList.txt. This is the part where it tries to find kindr, rovios other dependency. You will need to also clone the kalibr repo, cd to your src dir and go `git clone https://github.com/ethz-asl/kindr.git`

FPSychotic, 2018-02-12:
Please, dont missunderstand me, maybe my english level is not enough, didn't want mean that is not enough effort in this software,and less when is open source,just that is needed too much skills in linux and enviroment. A lot more than install other ros packages via apt-get. The main problem is my lack of knowledge. Apologies, even know I dont know if I explained properly

SrijaneeBiswas, 2018-02-12:
Hi Zach,

I have cloned the kindr repo. catkin build still fails with the same error message as above. Any ideas?

Thanks,
Srijanee

fmina, 2018-02-12:
I think the layout of your workspace is not correct. you need to change it to `merged` instead of `linked`. Check catkin tools webpage to see how to do it.

SrijaneeBiswas, 2018-02-13:
Hi Marco and Zach,

Thank you so much for your help. I got ROVIO to work on the Euroc dataset. Can you please guide me regarding what changes do I need to make to run this using my own real-time camera data?

Regards,
Srijanee

marco-tranzatto, 2018-02-14:
Hi,

@FPSychotic it's okay, I'm sorry if I misunderstood your comment.

@SrijaneeBiswas have a look here: [http://grauonline.de/wordpress/?page_id=1951](http://grauonline.de/wordpress/?page_id=1951)

Regards,
Marco.

FPSychotic, 2018-02-15:
Please, dont apologize it was my fault and thank you for the link, I check naw

yinyanpeng, 2018-03-10:
Hi,@ SrijaneeBiswas  This version supports ros kinetic?

yinyanpeng, 2018-03-10:
@SrijaneeBiswas  .This version supports ros kinetic?

mramezani64, 2018-08-02:
Hi,
Is there anyone know why I got the error below? I try to run rovio but I am stuck at the stage below. Any help would be appreciated.

[image]

ZacharyTaylor, 2018-08-02:
@mramezani64  see issue #171 as you have the same problem as was resolved there.

mramezani64, 2018-08-02:
Thank you @ZacharyTaylor  . It worked.

closed 2018-08-06 by bloesch