ethz-asl/rovio #152: Clarification on camera/world coordinates

swdeedee, 2018-02-27:
Hi,

I've setup rovio outside of a ROS environment, calibrated a camera and IMU to test on my own datasets, and everything is working smoothly so far.
Now I would like to use the output of the filter in my own OpenGL scene (like the built-in visualization scene), but I can't figure out the used coordinate frames. I've read the wiki and various tickets, but the output I get does not make sense to me.
I'm attempting to use the filter's state WrWC and qCW to produce a view matrix for my scene (eg. get a first person view of the rovio camera), which uses a "traditional" Y up / -Z forward. I understand that the rovio camera uses Y down / Z forward coordinates.
Now the output of the filter does not make sense to me; for example, a dataset with a motion translating toward the left (camera point of view) increases the Y (2nd) component of the WrWC vector, and a dataset moving forward increases the X (1st) component of WrWC. Also, if I setup a view matrix with rovio::InitTransform, my ground plane appears on the right of my camera view.

I'm probably missing something, but I can't figure it out.

Also, are the translations using "real world" units (eg. meters) ? This would make sense for ground truth comparison, but I would like to make sure.

Thanks for any clarification.

ZacharyTaylor, 2018-02-27:
The transform distances are in meters. For transform conventions rovio uses a right hand coordinate frame where as many opengl systems use a left hand frame so this could be your issue. Otherwise this could be a problem with active vs passive frames. Unfortunately there are so many competing ways of representing rotations that interfacing a new library which has its own conventions usually turns into quite a bit of trial and error.

swdeedee, 2018-02-28:
After poking a bit with the RovioScene, I realized that the world coordinates are expressed Z up and XY on the ground plane (right hand), which now makes sense with the filter's output.
Thanks for you help.

closed 2018-02-28 by swdeedee