ethz-asl/rovio #154: imu intrinsics & cam/imu extrinsics

han-so1omon, 2018-03-01:
Hi, I've ported rovio to px4 and have some questions:

For imu intrinsics:
Does rovio make its own estimates of imu noise densities and random walks?
Does rovio require imu measurements scaled (and adjusted for nonlinearities) to meters?

For cam/imu extrinsics:
Does rovio require cam/imu time and axis sync?

bloesch, 2018-03-02:
Concerning your questions:
- you have to give rovio noise densities in the cfg file, it does not estimate them
- rovio assume calibrated imu measurements. accelerations in m/s^2, rotational rates in rad/s
- timestamps of imu and cam need to have a common reference time
- you can use radial-tangential or equidistant camera models

han-so1omon, 2018-03-02:
What is the name of noise densities and random walks parameter?
In Kalibr it is `<accelerometer|gyroscope>_<noise_density|random_walk>, but there is no such variable in the `rovio.info` file nor is there a place for reference to an `imu.yaml` file. Is it a subset of the PredictionNoise parameters?

bloesch, 2018-03-03:
the parameters are under Prediction.PredictionNoise. you will have to change vel_* (acc noise), acb_* (acc bias noise), att_* (gyro noise), gyb_* (gyro bias noise). be sure to pay attention to the units.

the equidistant model is often used to calibrate fisheye lenses and is equivalent to the one employed in opencv fisheye.

han-so1omon, 2018-03-04:
Can't figure out why units of vel_ (acc noise) and att_ (gyro noise) include an extra time term (i.e. m^2/s^3 vs m^2/s^4 & rad^2/s vs rad^2/s^2). Total RMS noise is given in m/s^2 and rad/s. Is there some divide by bandwidth frequency term that I'm missing?

bloesch, 2018-03-05:
these are noise densities (squared). E.g. for the gyro you often get something like rad/s/sqrt(hz) from the manufacturer (sometime also with different units). If this is squared this gives rad^2/s, which is what rovio requires.

closed 2018-03-08 by han-so1omon