ethz-asl/rovio #157: Running ROVIO in Gazebo

RyanMorris95, 2018-03-05:
Hi,
First of all, thank you for open sourcing such a great project.

I have been trying to run the rovio_node through gazebo and haven't been able to get any output from rovio.  I have a python file that takes camera images, converts to grayscale, and then publishes to rovio.  By running rqt_graph I can see that camera and imu is being sent to rovio.  The rovio node runs and the camera opengl scene opens, but doesn't display any feature points and no data is published from the node.

Here is my python wrapper class:
`#!/usr/bin/env python
import sys, select, termios, tty

import cv2
import rospy
import matplotlib.pyplot as plt
from sensor_msgs.msg import Image, PointCloud2, Imu
from nav_msgs.msg import Odometry  # for wheel and visual odometry
from geometry_msgs.msg import PoseWithCovarianceStamped
from cv_bridge import CvBridge, CvBridgeError

class RovioWrapper(object):
    """
    All this class does take the camera feeds, undistorts them, and then publishes
    to the ROVIO topis.
    """
    def __init__(self):
        self.camL_sub = rospy.Subscriber('/rogue_two/camera_stereo_left/image_raw', Image, self.imageL_rectify)
        self.camL_pub = rospy.Publisher('/cam0/image_raw', Image)
        self.bridge = CvBridge()

    def imageL_rectify(self, image_l):
        # rectify code will go here on the real robot
        image_l_cv = self.bridge.imgmsg_to_cv2(image_l, desired_encoding="rgb8")
        image_l_cv = cv2.cvtColor(image_l_cv, cv2.COLOR_RGB2GRAY)
        image_l = self.bridge.cv2_to_imgmsg(image_l_cv)
        self.camL_pub.publish(image_l)

if __name__ == '__main__':
    rospy.init_node('rovio_wrapper')
    rovio_wrapper = RovioWrapper()
    rospy.spin()
`

Here is my launch file:
`<?xml version="1.0" encoding="UTF-8"?> 
<launch>
  <node pkg="rovio" type="rovio_node" name="rovio" output="screen">
  <param name="filter_config" value="$(find rovio)/cfg/rovio.info"/>
  <param name="camera0_config" value="$(find rovio)/cfg/gazebo_cam.yaml"/>
  <remap from="/imu0" to="imu"/>
  </node>
</launch>
`

I am not getting any errors from the rovio node, however, I do get a warning that says, "included measurements at time 0 before safe Time."  Is that a critical issue? 

Any ideas what the issue could be?

Thank you for your assistance,
Ryan

bloesch, 2018-03-05:
Can you double check your timestamps? The warning your are receiving is sent when some of the measurements rovio sees are too far in the past.

Please double check all timestamps (including the first one). Also timestamps of IMU and images should be w.r.t. a common reference.

szebedy, 2018-05-13:
Hi @bloesch ,

Big up for the great project!

I'm having the exact same issue (image and imu data coming from Gazebo), except I get no warnings from rovio. It passes every test successfully but then there is no change on the opengl scene, and no data is published from rovio.

Any suggestions on how to proceed?

Here is my launch file:
```XML
<?xml version="1.0" encoding="UTF-8"?> 
<launch>
  <node name="image_proc" type="image_proc" pkg="image_proc" output="screen">
    <remap from="/image_raw" to="/camera/rgb/image_rect_color"/>
    <remap from="/camera_info" to="/camera/rgb/camera_info"/>
  </node>
  <node pkg="rovio" type="rovio_node" name="rovio" output="screen">
    <param name="filter_config" value="$(find rovio)/cfg/rovio.info"/>
    <param name="camera0_config" value="$(find rovio)/cfg/euroc_cam0.yaml"/>
    <!-- param name="camera1_config" value="$(find rovio)/cfg/euroc_cam1.yaml"/ -->
    <remap from="/imu0" to="/mavros/imu"/>
    <remap from="/cam0/image_raw" to="/image_mono"/>
  </node>
</launch>
```

LZMHIT, 2018-05-14:
Hi,
   Thank you for the Open source.
   I have some problems when I use the gazebo to run Rovio, I use the vi_sensor which comes form the model in rotos_simulator, I mount it on another flying vehicle with towards downward, I know the coordinate of the vi_sensor/imu is different from the set in the picture in your [homepage](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets), but I change it in the .xacro file, Then I modify the cam.yaml file as well as the rovio_filter.info,  however when I run rovio, I found it dosen't work at all, it drift soon. Could you please tell me What is the matter with my work? Thank you sincerely!

szebedy, 2018-05-20:
@bloesch @RyanMorris95, I've found the solution to our problem. You've gotta `<remap from="/imu0" to="imu/data"/>` not just simply  `to="imu"`... I find this kinda counter intuitive since the input is simply called imu0 which would suggest I have to remap to imu, but who am I to judge...

bloesch, 2018-05-29:
@szebedy thanks for the input! Having some warnings on ROVIO's side for checking the data would probably also be helpful.

@LZMHIT: you have a different issue, but I definitely think it has to do with the calibration.