ethz-asl/rovio #158: something about the paper theory

zhuzunjie17, 2018-03-07:
@bloesch 
I want to know more theory about the rovio paper and i heard that your doctoral thesis provide more detailed formula. But i can't download your doctoral thesis in ETH website, could you send me your doctoral thesis? Thanks.

bloesch, 2018-03-07:
If you have not had a look at the IJRR paper this might give some further information. Otherwise, the thesis does not include additional information as it is only a collection of papers. It was withheld due to unpublished work but will be available shortly.

zhuzunjie17, 2018-03-07:
Thanks for your reply, I have read the IJRR paper and another EKF-based ROVIO paper. However i still can not full understand the datails of the IEKF-based method some information did not metioned on the paper(if so). So i wanna know if there are any paper you published which provide more information, or details only showed in code?
another thing is that, could you share your email to me, the email in IJRR paper is invalid.

bloesch, 2018-03-07:
For more background understanding Barfoot's book is a great read (http://asrl.utias.utoronto.ca/~tdb/bib/barfoot_ser17.pdf). Otherwise you can also try the primer I wrote but this is rather difficult to understand (https://arxiv.org/pdf/1606.05285.pdf). The email address in the published IJRR is correct (http://journals.sagepub.com/doi/full/10.1177/0278364917728574).

zhuzunjie17, 2018-03-07:
Thanks for your advice, its helpful.

I still cant send email to the address, strange.
[image]

bloesch, 2018-03-07:
this is the old address, please see the address in the paper on the sagepub website.

closed 2018-03-08 by zhuzunjie17