ethz-asl/rovio #162: IMU bias and extrinsics do not converge

JJQ7, 2018-03-17:
Hi all,
I'm using Ubuntu 16.04 and ROS Kinetic, everything properly worked when i run rovio with the provided test data.
Then I changed some imu related covariance parameters according to the device values provided by the manufacturer and tried to run with my device (IMU: ICM 20602, and a hardware synchronized camera).
Rovio passes the tests of no motion and little motion. However, when test with large motion, the outputs have a large difference with the ground truth. And sometimes, rovio diverges.
I adjust some parameters to make rovio performs better. But it seems that the parameters can not guarantee similar performance with different data sets (recored by the same device).
I observe that the imu bias and imu-camera extrinsics always changes rapidly under different parameters i have tried. I really hope someone could give me some advices. 
Thank you!

JJQ7, 2018-03-17:
[image]

JJQ7, 2018-03-17:
Aditionally, it seems that the feathers are not tracked correctly when the camera shakes.

JJQ7, 2018-03-17:
#70 
#80 
#81

JJQ7, 2018-03-26:
I notice something interesting, if I change covariance parameter of attitude prediction, the size of the trajectory will change. And The effect of tracking landmarks seems to have a close relationship with this parameter. However, I still can't get an  ideal result. Could someone give me some advices on these parameters? Thank you !

bloesch, 2018-03-30:
Something does indeed seem off. In my experience the gyroscope biases typically converge quite reliably (but then again the scale in your plots is quite small). Also a accelerometer bias of 1m/s^2 seems to large. How do the extrinsics look like? Do you use any calibration for extrinsics or do you pass identity?

There is some description on the parameters in the wiki (https://github.com/ethz-asl/rovio/wiki/Configuration). I am not familiar with the IMU you mentioned but in my experience the default noise parameters are typically too low (the IMU employed in the euroc dataset is rather good). So you should probably slightly increase the values (a factor 10-1000 because the unit are squared).

JJQ7, 2018-04-01:
Test 1
[image]
[image]
[image]
[image]

JJQ7, 2018-04-01:
Test 2 (same device, same parameters)
[image]
[image]
[image]

JJQ7, 2018-04-01:
Thanks for your advice. And sorry for my late reply. My computer is broken so everything is lost...
According to your suggestion, I adjusted the parameters, and conducted two tests (using same device and some parameters. velocity is recorded by Qualisys motion capture system). However, there is still some problems about extrinsic and bias. It seems that rovio is trying to attribute measurement error the the extrinsic and bias.(especially in bias)

JJQ7, 2018-04-01:
I will try to give a fixed extrinsic provided by Kalibr and upload the results in the future.

JJQ7, 2018-04-04:
[image]

bloesch, 2018-04-05:
It's difficult to say how good the biases/extrinsics should converge. Most of your plots for these values look okay. The steps in the last one (just above) are a weird though. What setup did you use.

Your comment concerning "attributing measurement errors to biases" is valid. The filter decides online what states to correct given the measurement error and this is done based on the estimated uncertainty of all states. So if your bias states (including extrinsics) do not converge well then measurement errors will affected them more. But all of this is very dependent on how you tune the covariances and just taking the covariance specifications from the device is unfortunately rarely enough (due to inaccuracies in the modeling). If you feel the bias states are being affected to strongly you can reduce their prediction noise (at the cost of some robustness w.r.t. bad initial convergence), reduce their initial uncertainty (at the cost of wrong initial guesses), or fix the extrinsics.

JJQ7, 2018-04-05:
yes! I found even in the pubic data (euroc MH_05), the bias provided by rovio is not a fixed value although the final trajectory is close to the ground truth. This really bothers me for a long time. And some divergences come up in euroc data. Maybe some minor adjustments can improve the performance of rovio.

In the last plot (just above),  the same device was used. The data is record on a quadrotor. The quadrotor moves along the trajectory of a square and stays at the four vertices for a certain period of time. The extrinsics change when quadrotor moves.