ethz-asl/rovio #166: Doctoral thesis request

vonzy, 2018-04-09:
Hey Bloesch, it's a pity that I have only recently seen your excellent work. I am a first-grade post-graduate and want to implement robust monocular visual-inertial state estimation on android devices. I have tried OKVIS, VINS-Mono and ORB-SLAM2, neither of them can deal with huge motion. Feature tracking failed easily when it comes to huge motion.
I wonder maybe it's due to IMU measurements are not made full use of. In your paper, IMU measurements are used to help feature tracking, and the effect is quite stable and robust as your video shows. So recently I am reading your code and papers, but there are still a lot of confusion about details. I wonder could you please give me your doctoral thesis，So I can understand more about the code. My email address is von9604@gmail.com. Thank you very much!

Yours sincerely,
Vonzy

vonzy, 2018-04-09:
@bloesch

salahkhan94, 2018-07-19:
Hi Bloesch,
I too would like to learn your VIO implementation in depth, I still have a lot of confusion about many things even after going through your Code and Papers. I'm hoping your Doctoral thesis would help me get a much deeper understanding of ROVIO. E-mail : salahuddin.khan.1994@gmail.com. 
If there is any other resource that could help understand Rovio better please do share 'em.

Regards,
Sal.