ethz-asl/rovio #167: Questions about Visual inertial odometry development

Ilyaprok, 2018-04-11:
Hello! I am a student from Russia of MGTU Nosov from Magnitogorsk.Topic of my graduation work is Visual Inertial Odometry.  I recently read both your publications (2015, 2017) of ROVIO and looked code. I very like your approach with tight fusion of IMU and Video.
Now I implemented Square Root Central Difference Kalman Filter (from publications Rudolph van der Merwe and Eric A. Wan) with your approach (robocentric representation, bearing vectors and inverse depth of features in state, your kinematics model) but measurement model is not your (I use vector of pixel coordinates instead error intensity). I implemented 2 versions of this filter (conventional SRCDKF and SRCDKF with lie algebra from publications Hetzberg). But second version (with lie algebra, with minimal representation of covariance matrix) work worse than conventional filter with full covariance matrix. I don't understand why? Both versions diverge after some time or turn to NaN. I don't understand why?
I seen in lightweight filtering code with tests UKF with algebra lie (boxplus and boxminus), but I not found any results. Why did you take as the basis the EKF and not the UKF? In publications Rudolph van der Merwe and Eric A. Wan sigma points kalman filters are better work than EKF...
Sorry for my English :)
Thank you in advance :)

bloesch, 2018-05-29:
Replacing the intensity error with traditional reprojection errors: this is totally a valid approach and we have been experimenting around with it as well. In many settings the performance is comparable to the implementation with the photometric error.

Lie algebra: this usually leads to better accuracy (not only from a theoretical point of view). The fact that both of your implementations evtl drift seems to indicate that something else is still wrong with your framework.

UKF vs EKF: if nonlinearities are small (good linearisation point, small time increment) then there is no significant advantage of the UKF over the EKF. However, the UKF is computationally more expensive (at least in our implementation) and would struggle to support realtime performance.

Ilyaprok, 2018-05-29:
Thank you very much for answer!
Yes you are right. I found some errors in my implementation with lie algebra.  Now SRCDKF with lie algebra better than without.

closed 2018-08-06 by bloesch