ethz-asl/rovio #170: ROVIO on Raspberry Pi

raoulj, 2018-05-01:
I'm trying to get ROVIO on a system of connected Raspberry Pis.  Currently I have the ROVIO node running ROVIO on my desktop (cameras on a series of Pi Zero W's), but I'll test it on a Pi3 once I can get it to work robustly.

I've been having trouble with my feature tracking.  Below are two pictures, one taken about 5 seconds after the ROVIO node launches, the other 5 seconds after that.

[image]

[image]

I originally thought that this was caused by poor calibration (the convergence of the feature points to a single location on the screen reminded me of negative radial distortion), except I've recalibrated several times now to no avail.  The only other way that I see my datasource differing from EuRoC is that my inertial sensor can't do much better than 80 samples a second, whereas EuRoC is around 200.

Any pointers would be greatly appreciated, thank you!

raoulj, 2018-05-02:
I realize that my issue is likely due to a lack of proper calibration.  Going through the Kalibr process now and will report back results.

mhkabir, 2018-05-02:
Rovio is runs in a single compute-intensive thread and is extremely sensitive to per-frame timing. The RPi does not have a high enough single-core clock speed to run Rovio without diverging on motions.

chrissunny94, 2018-05-11:
Is there any SoC that's capable of performing rovio on-board ?

ZacharyTaylor, 2018-05-12:
I have run it on an up-board and an Intel Euclid. I believe one of the better Intel compute sticks would also work. However, running it on these systems requires you to tune down the patch sizes, number of tracked points and use a fairly low framerate, all of which can hurt robustness.

aseyfi, 2021-04-13:
@mhkabir , what is the minimum recommended resolution? Do you think a 320x240 resolution would produce results that are accurate enough for drone navigation?