ethz-asl/rovio #174: Frame conversion issues.

rasmaster, 2018-05-18:
Hey, 

I have been having some difficulties finding the right transformation from my IMU frame to camera, 
and i'm starting to believe that im doing something wrong elsewhere. 
When i read my raw imu data, through mavros, it appears that it has a left-handed frame (x pointing backwards, y right and z down). Since the Rovio camera frame is right handed, i manually convert my imu frame through a ros script, so that it is right handed (same frame as EuRoc dataset x up, y right and z forwards). 

However, when I give the rotation to camera frame in the rovio.info, my pitch seems to be inverted when i look at the attitude indicator in the Rovio display. I've tried different imu frames with different IMU to cam transforms in the rovio.info and used the attitude controller to validate whether or not it is the right transform, but everything turns out with (at best) either roll or pitch inverted. When i run kalibr imu/cam, the T_ic outputs the expected rotation to the conventional camera frame. 

I tried with a left-handed coordinate IMU frame and tried different transformation, and managed to have a combination where the attitude indicator appears to be working as it should. This is also the combination that has provided me with the best results so far. In this combination, my IMU frame is x down, y left and z backwards and the rotation is x=0, y=0, z=-0.7071 x=-0.7071. With this combination i have some troubles rovio struggles with rotations, especially pitch, but better than what i've achieved with right handed frames. 

Is there something i am missing? 

My system: im using a pixhawk autopilot and raspberry pi cam, and i run the algorithm on my pc. 
The data isn't hardware synchronized. I've made at attempt to solve it by software, which seems to be working sort of ok.

Thank you in advance to anyone who might answer my questions.

bloesch, 2018-05-29:
Having IMU readings in a left handed frame is very special, are you sure that is right? Also, the fact that you get exactly one axis inverted speaks against the left-handed CF fix you make. Also, did you run kalibr with the original IMU readings?

renielse, 2018-05-29:
I actually don't read my raw  imu readings, i read them through Mavros. My pixhawk is connected to a RPi which uses MAVproxy to relay mavlink messages. So the IMU readings is actually being send through both Mavproxy and mavros ... maybe something goes wrong there. But im certain that the readings that i get from mavros are left-handed, and the angular velocities are also opposite to the right-hand rule. 

About the inverted axis, that only happened with right handed frames. I don't think i explained that properly. I never achieved to make the attitude indicator behave properly with right handed-frames, either pitch or roll would be inverted. With left handed frames it behaves as it should.

Another thing worth mentioning is that with right-handed frames the online calibrations provides some extreme extrinsics (1.5 m translation on z and y axis), but with left-handed frames they converge quite nice. 

I did run kalibr with original imu readings from mavros. The rotation given by T_ic suggest that the camera frame is the exact opposite of the conventional camera frame (x left, y up and z backwards).