ethz-asl/rovio #175: How to align groundtruth pose and rovio odometry？

wlwsjl, 2018-05-18:
[image]

JJQ7, 2018-05-24:
You can start the rovio with the Init State of groundtruth, especially for the yaw angle from init attitude quaterion. By the way, what device are you using? It seems that you'v got a desirable result.

bloesch, 2018-05-29:
It depends what you want the alignment for. If you only need the alignment for evaluation then post processing is going to be the easiest way. If you require alignment during runtime then it might get a bit trickier. YOu cannot align to arbitrary pose as rovio will do an internal gravity alignment since it required gravity to be along the z-axis. Again the easiest might be to do it in a postprocessing step with a fixed alignment that you compute after rovio has initialised.

wlwsjl, 2018-06-04:
Thank you @bloesch . I've done the postprocessing and get good results.
[image]

wlwsjl, 2018-06-04:
[trajectory_toolkit](https://github.com/ethz-asl/trajectory_toolkit)

cumtchenchang, 2018-06-22:
@wlwsji  How to get the trajectory of results?

tfe104270, 2018-07-23:
I found a package that did some vslam comparison and output rovio result and groundtruth.
https://github.com/nicolov/vslam_evaluation