ethz-asl/rovio #176: How to run rovio with RealSense ZR300?

zabuton0807, 2018-05-21:
Hi all,
I am trying to run rovio with RealSense ZR300. It seems that tf frame is published correctly when moving slowly. However, when I move it violently, tf frame diverges. I use following configuration files.

camera.yaml: https://github.com/ethz-asl/mav_dji_ros_interface/blob/master/dji_interface/cfg/raven/rovio_cam0.yaml

filter.info:https://github.com/ethz-asl/mav_dji_ros_interface/blob/master/dji_interface/cfg/raven/rovio_filter.info

Is there any other necessary operation? Could someone teach me? Thank you.

kungfrank, 2018-06-04:
Hi,
I am using ZR300 and same configuration files too. But my terminal just stop at 
`[FilterBase::addUpdateMeas] Warning: included measurements at time 1528122046.37 before safeTime 1528122046.37`
and there are no any window show out. Have you face this before?

helenol, 2018-06-05:
You should probably calibrate your own camera using kalibr, especially for the intrinsics, and turn online extrinsic estimation to true: `doVECalibration false;` in the info file and see if that improves stuff. In general we found that for the realsense we had to go a few orders of magnitude up with the IMU noise, so that is something to try. Make sure you are also using the ASL drivers, https://github.com/ethz-asl/realsense , which fix tons of timesync issues and so on (and still support the ZR300).
In general we found that the ZR300 never worked particularly well with crazy motions, since the IMU is just very low quality, so there's definitely a limit to how good it can get.

zabuton0807, 2018-06-06:
@helenol Thank you for your comment. I can run rovio with ZR300 now. However, I faced another problem. Would you please check it if you do not mind? 

https://github.com/ethz-asl/rovio/issues/179

Thank you.

closed 2018-06-06 by zabuton0807

jingchen93, 2018-12-23:
> You should probably calibrate your own camera using kalibr, especially for the intrinsics, and turn online extrinsic estimation to true: `doVECalibration false;` in the info file and see if that [...]

> You should probably calibrate your own camera using kalibr, especially for the intrinsics, and turn online extrinsic estimation to true: `doVECalibration false;` in the info file and see if that [...]

Hello, may I know what are the required packages/dependencies for the ASL drivers to work. Currently I am facing an issue when building the ASL version of realsense2_camera packages. 
{
In file included from /home/jc93/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/base_realsense_node.h:6:0,
                 from /home/jc93/catkin_ws/src/realsense/realsense2_camera/src/../include/rs415_node.h:6,
                 from /home/jc93/catkin_ws/src/realsense/realsense2_camera/src/rs415_node.cpp:1:
/home/jc93/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/../include/mavros_timestamp.h: In member function ‘void mavros_trigger::MavrosTrigger<t_channel_id, t_cache>::start()’:
/home/c93/catkin_ws/src/realsense/realsense2_camera/src/../include/../include/../include/mavros_timestamp.h:133:19: error: ‘mavros_msgs::CommandTriggerControl::Request {aka struct mavros_msgs::CommandTriggerControlRequest_<std::allocator<void> >}’ has no member named ‘integration_time’
       req.request.integration_time = 1.0;
}
Currently I have isntalled ethz-asl version of mavros and mav_comm and original librealsense_master from Intel.
I am looking forward for the feedback, thanks!