ethz-asl/rovio #181: Installation on odroid

lguanrui, 2018-06-07:
Hey guys, 

I'm trying to install ROVIO on Odroid XU4 but I constantly got killed during the catkin build. I have tried with catkin build -j2 or j1 but neither of them works. Here is the error I got from the catkin build: 
[image]

Hope to hear some help soon
Regards
lguanrui

mhkabir, 2018-06-08:
Try enabling swap space.

lguanrui, 2018-06-08:
Hey mhkabir, 

Thanks for the reply. The swap space ideas works in a way. I assigned 2GB swap space for the odroid and it still got killed when building the rovio_rosbag_node. However it successfully built without building rovio_rosbag_node. I tried to run the rovio_node with my camera but it gave me this following error, do you have any idea? 
[image]

Hope to hear you soon
Thanks!!
lguanrui

lguanrui, 2018-06-08:
The package successfully built after I assigned 4GB swap space!! but still it gave me the above error.

lguanrui, 2018-06-12:
Thank you so much!! It did the trick!

closed 2018-06-12 by lguanrui