ethz-asl/rovio #182: Hardware specifications for real time odometry

connorsoohoo, 2018-06-12:
Can someone give what their exact hardware specifications were and the configuration used to analyze camera images in real time?

bloesch, 2018-08-06:
hardware specification for the euroc dataset can be found online (https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets). The default rovio settings should run fairly well on this dataset.

habib-bolourchi, 2018-08-29:
Hey @bloesch,
I already bought and optor vi_sensor ( it is the github: https://github.com/optor-vis/vi-stereo-v1 ) it has a ros package and I can connect it to your code but the problem is that it drifts I think it is about the frame speed. I am totally new to the hardware. Should I calibrate it with Kalibr or should I use its Ros package. If I should use their ROS package how should I change the speed and what is the number of fps I should put in it.
I would appreciate it if you help me with this issue I am really stalked.

habib-bolourchi, 2018-08-30:
Thank you, I think I use an imu via arduino , but would you please tell me which number should put as a frame rate in fps for camera and imu? so then I can get just the images from optor since it has global shutter cameras and if I understood well from what you said, it seems the cameras are not the problem.

ZacharyTaylor, 2018-08-30:
The camera images shouldn't be an issue, however rovio needs the relative timing between when an IMU reading occurred and an image was taken to be given quite accurately. As the sensor only stamps images on arrival on the pc you will need to factor in the unknown internal imaging processing time and time for the images to be transferred. This can be very difficult to do accurately which is why we generally aim for using sensors that provide hardware timestamping. I have successfully run rovio on cameras with framerates of 12, 20, 30 and 60 fps and IMUs with rates of 100, 200 and 250 Hz. Our current setup has images at 20 fps and IMU at 200 Hz, but it does not matter too much.

mzahana, 2018-10-04:
@ZacharyTaylor If possible, could you please share what hardware you use to get good results with ROVIO?

Thanks.

ZacharyTaylor, 2018-10-04:
I wrote up some information on our VI sensor setups here https://github.com/ethz-asl/mav_tools_public/wiki/Visual-Inertial-Sensors

We generally use the skybotics VI sensor, but due to not being able to get them anymore are transitioning to a setup triggering a camera from a microcontroller.

I should also note while cheaper imus work, we usually go with the adis 16448 due to it's small bias which helps robustness

mzahana, 2018-10-04:
I would also like to mention that I have an arduino and MPU6050 with the intention to build a complete VI rig.

I came across this tutorial which explains how to use arduino to sync MPU6050 with a global shutter camera. However, I don't have the camera mentioned in that tutorial, but I have the uEye 1226LE.

habib-Boloorchi, 2018-11-05:
Hey @ZacharyTaylor,
Thank you for your help, I connected a trigger of ArduCam which is a global shutter  using this code : https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/ROS/arducam_usb2_ros/src/arducam_trigger_ros_node.py
it works well and I connected them to a raspberry pi as you see n the picture I use an external shutter
[image]
I want to use raspberry just as a data collector but my problem is I want to get data from IMU (MPU_6050) and I thought maybe I use this code: https://github.com/chrisspen/ros_mpu6050_node but I am not sure how to synch them without Arduino. 
would you tell me if it is feasible to do this or it can be just done with the clock of an Arduino?
I also appreciate your help in telling me how to make them work together.