ethz-asl/rovio #184: Running on EuRoC?

peter-zhao, 2018-06-19:
Hi,

I tried to run rovio in stereo mode on EuRoC benchmark.  Since there exist a set of default config files (e.g. euroc_cam0 & 1.yaml, rovio.info) in the repo, I assume one would be recommended to utilize them for EuRoC benchmark.

However I ended up with obvious divergence when running stereo rovio with these default config files.  For example, it diverges after 10 sec or so on MH_01_easy.  I wonder is there any "correct" config files for EuRoC, since it is one of the go-to benchmark for VSLAM/VIN folks?

Thanks for any feedback or suggestion!

MaxGamerProPlayer, 2018-06-25:
👍

helenol, 2018-06-26:
Run mono rovio instead; this is the stabler, better version. ;) This is also what we run on-board, etc.

peter-zhao, 2018-06-26:
Hi Helen,

Thank you for the response!  Since I am trying to compare different stereo VINS systems, rolling back to mono rovio doesn't really fit (although I know that mono version is more mature).
Besides, in the rovio journal paper published on IJRR 17, there is comparison between mono rovio, stereo rovio and okvis on EuRoC sequence.  I wonder if the current rovio impl can re-produce the similar level of performance with stereo mode?  Or do I have to use the ijjrInitial release that was created 2 years ago?

bloesch, 2018-06-28:
There are various issues with the stereo version and the use of the mono version is definitely more recommended (due to various fixes it might also give you better results than the stereo version from two years). You can roll back to ijjrInitial but you might also get problems with dependencies.

However, an issue with evaluating stereo is the sensitive dependency of the performance w.r.t. the quality of the calibration. For the IJRR publication, if I remember right, we had tuned the parameters such that they allow bad initial guesses of the extrinsics to converge well given enough excitation. Now in a setup where the extrinsics are well-known you could get better performance by either disabling the estimation of the extrinsics or by reducing the corresponding covariance parameters.