ethz-asl/rovio #185: Ubuntu 16.04 Crashes on UP Board

mhkabir, 2018-06-22:
Running the Rovio rosbag node, which crashes on parsing the first message : 

```
Thread 1 "rovio_rosbag_lo" received signal SIGSEGV, Segmentation fault.
0x00000000004f8bae in void Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<double, -1, 1, 0, -1, 1>, Eigen::Product<Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, Eigen::Matrix<double, -1, -1, 0, -1, -1> const>, Eigen::Matrix<double, 2, 1, 0, 2, 1>, 1>, Eigen::internal::assign_op<double> >(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, Eigen::Product<Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, Eigen::Matrix<double, -1, -1, 0, -1, -1> const>, Eigen::Matrix<double, 2, 1, 0, 2, 1>, 1> const&, Eigen::internal::assign_op<double> const&) ()
(gdb) bt
#0  0x00000000004f8bae in void Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<double, -1, 1, 0, -1, 1>, Eigen::Product<Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, Eigen::Matrix<double, -1, -1, 0, -1, -1> const>, Eigen::Matrix<double, 2, 1, 0, 2, 1>, 1>, Eigen::internal::assign_op<double> >(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, Eigen::Product<Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, Eigen::Matrix<double, -1, -1, 0, -1, -1> const>, Eigen::Matrix<double, 2, 1, 0, 2, 1>, 1> const&, Eigen::internal::assign_op<double> const&)
    ()
#1  0x000000000051aeb6 in LWF::Update<rovio::ImgInnovation<rovio::State<20u, 3, 4, 1, 0> >, rovio::FilterState<20u, 3, 4, 1, 0>, rovio::ImgUpdateMeas<rovio::State<20u, 3, 4, 1, 0> >, rovio::ImgUpdateNoise<rovio::State<20u, 3, 4, 1, 0> >, rovio::ImgOutlierDetection<rovio::State<20u, 3, 4, 1, 0> >, false>::performUpdateEKF(rovio::FilterState<20u, 3, 4, 1, 0>&, rovio::ImgUpdateMeas<rovio::State<20u, 3, 4, 1, 0> > const&) ()
#2  0x000000000057fd42 in LWF::FilterBase<rovio::ImuPrediction<rovio::FilterState<20u, 3, 4, 1, 0> >, rovio::ImgUpdate<rovio::FilterState<20u, 3, 4, 1, 0> >, rovio::PoseUpdate<rovio::FilterState<20u, 3, 4, 1, 0>, -1, -1>, rovio::VelocityUpdate<rovio::FilterState<20u, 3, 4, 1, 0> > >::update(rovio::FilterState<20u, 3, 4, 1, 0>&, double const&) ()
#3  0x00000000005802a3 in LWF::FilterBase<rovio::ImuPrediction<rovio::FilterState<20u, 3, 4, 1, 0> >, rovio::ImgUpdate<rovio::FilterState<20u, 3, 4, 1, 0> >, rovio::PoseUpdate<rovio::FilterState<20u, 3, 4, 1, 0>, -1, -1>, rovio::VelocityUpdate<rovio::FilterState<20u, 3, 4, 1, 0> > >::updateSafe(double const*) ()
#4  0x0000000000580808 in rovio::RovioNode<rovio::RovioFilter<rovio::FilterState<20u, 3, 4, 1, 0> > >::updateAndPublish() ()
#5  0x0000000000487e67 in main ()
```

I tried adding `add_definitions(-DEIGEN_DONT_ALIGN_STATICALLY)` to see if it was an alignment issue, but it didn't help. Any ideas @nicolov @ZacharyTaylor @bloesch? Is there something I can do to narrow this down a bit more?

mhkabir, 2018-06-22:
Looks like it's an Eigen bug. Manually linking to Eigen 3.3.4 pulled from source fixes the crashes.

closed 2018-06-22 by mhkabir

MaxGamerProPlayer, 2018-06-25:
👍