ethz-asl/rovio #189: Rovio with 2 timestamped but not synchronised cameras

mhkabir, 2018-07-06:
Hi @bloesch,

Is it feasible to run Rovio with 2 independently running cameras (not in stereo config, not overlapping) which are timestamped against the same base? If so, how would the filter logic have to be modified to allow asynchronous image updates?

Thanks for the help!

mhkabir, 2018-07-10:
@helenol @ZacharyTaylor @nicolov If you guys have any ideas, would be glad to hear them! :)

bloesch, 2018-08-06:
sorry for the delay. It depends whether the camera are synchronized. If yes, you should be able to use rovio in its native form. It should actually perform relatively well and most probably be better then with overlap (for some reasons having overlapping cameras results in a very settings sensitive framework).

If the cameras are not synced then this will get more difficult. I feel you would have to duplicate the update step. Somehow add two ImgUpdate<FILTERSTATE> to LWF::FilterBase such that you have two independent measurements.

closed 2023-07-17 by mhkabir