ethz-asl/rovio #191: Problem launching monocular-inertial Rovio. I can't see the tracker0 window.

odinhr, 2018-07-13:
Hi all,

I'm trying to launch Rovio with IMU and one camera. I have followed the kalibr calibration and after I have done the conversion to Rovio config (with command kalibr_rovio_config --cam file.yaml). But now, when I launch Rovio I can't see the Tracker0 window and my alghoritm doesn't work. However, if I play the euroc dataset it works perfect. I think my problem is in the parameters and info files format. Can anyone guide me to solve it ?

Thank you in advance.

odinhr, 2018-07-13:
Of course @ZacharyTaylor.

This is my rovio.info file: 

[rovio200hz.txt](https://github.com/ethz-asl/rovio/files/2192276/rovio200hz.txt)

and this is my .yaml file 

[rovio_cam20hz .txt](https://github.com/ethz-asl/rovio/files/2192277/rovio_cam20hz.txt)

Thanks for your attention. I think it could be the problem.

ZacharyTaylor, 2018-07-13:
There seems to be a couple of issues. The most major is
"Cameraa" should be "Camera0" and you seem to be missing the parameter maxFeatureFrames

I have attached a modified version with these changes 
[rovio200hz.txt](https://github.com/ethz-asl/rovio/files/2192416/rovio200hz.txt)
let me know if it works.

Also what IMU are you using as the current parameters in the prediction are for a very high quality imu with hardware timestamping.

odinhr, 2018-07-13:
It still doesn't work. Maybe, it could be a problem in my launch file. 

This is my launch file:

<?xml version="1.0" encoding="UTF-8"?>
<launch>
  <node pkg="rovio" type="rovio_node" name="rovio" output="screen">
  <param name="filter_config" value="$(find rovio)/cfg/rovio200.info"/>
  <param name="camera0_config" value="$(find rovio)/cfg/rovio_cam20hz.yaml"/>
  </node>
</launch>

What do you think about it ?

I'm using the imu output from the mavros. My hardware is composed of Navio2 + Raspberry Pi Model B (running PX4 firmware) and an oCam camera. Navio2 is composed by 2 IMU's ( MPU9250 and LSM9DS1). After that I use "rosrun topic_tools relay" for renaming the topics to /imu0 and /cam0/image_raw. If I only do the last step, it will work with your launch and parameters files, but the algorithm is always lost (due to the calibration).

Thank you for your collaboration.

ZacharyTaylor, 2018-07-13:
I feel you will have some significant issues getting rovio to work with the described setup.

The lowest powered hardware I have seen successfully run rovio was an upboard which is significantly more powerful than a RPi and even then we had to tune the frame rate, number of tracked features and other settings down to a point where it was not particularly robust.
The next issue is time syncing something rovio can be quite sensitive to. The oCam does not appear to have a trigger line so you are going to estimate the timing delay between taking an image and it arriving at the computer. How are you currently timestamping this data, as just stamping images on arrival is unlikely to be sufficiently accurate.

Also on an unrelated note you should use the remap parameter in the nodes to change topic names. I believe relaying topics causes the copying of the message which will be very memory heavy for raw images.

odinhr, 2018-07-16:
I'm using the RPi for getting IMU - data from navio. I have set the ROS_MASTER_URI on my PC and the oCam is connected there too. I've thought that maybe it could be the time syncronization. Therefore, I have use the image from the picamera too in some test, always processing on PC (i7 7th gen, 8GB, NVidia 920MX) . My idea is to run the algorithm on an auxiliar Odroid XU4 in a future. Can you recommend me some cameras with an easy syncronization?

Thank you so much @ZacharyTaylor

habib-bolourchi, 2018-07-23:
Hello @ZacharyTaylor
I bought a stereo camera that has the same features that your camera and imu has in euroC. this is the SDK and ROS file:   https://github.com/optor-vis/vi-stereo-v1/tree/master/optor_VI_Sensor_SDK_V1.0
but I am new in ROS and I don't know how to change the parameters to create the ROSbag that can work with ROVIO, I am kind of lost and I would appreciate it if could tell me what should I change and what I have to change.
and another question can you suggest me a book or article or even an online course that I can be as pro as you?

ZacharyTaylor, 2018-07-23:
Hardware synced cameras and IMU should hopefully solve most of your timesync issues. Though, I had a quick look at how their timestamping is done and it does strike me as a bit hacky https://github.com/optor-vis/vi-stereo-v1/blob/master/optor_VI_Sensor_SDK_V1.0/ROS/optor_stereo_visensor/src/optorimu.cpp#L224

As for help with ROS check out the following links:
* [Introductory Course](http://u.cs.biu.ac.il/~yehoshr1/89-685/)
* [Guided Tutorials](http://wiki.ros.org/ROS/Tutorials)
* [Documentation](http://wiki.ros.org/)
* [Cheat Sheet](http://www.clearpathrobotics.com/wp-content/uploads/2014/01/ROS-Cheat-Sheet-v1.01.pdf)

ZacharyTaylor, 2018-07-27:
It turns out there is a second person using the same sensor as you with rovio #192 
Because of this I took a closer look at the driver and found everything is just being stamped on arrival in software. There is a bit of a hacked together filter on the IMU stamps, but dispite the claims around this device you are not getting any hardware timestamping info. This will be an issue if you are trying to run rovio as it is quite sensitive incorrect timing information.