ethz-asl/rovio #195: Regarding covariance and noise tuning

indigomega, 2018-08-23:
Hey there,

I managed to get ROVIO running on my setup with a cheap IMU (px4). At first I was getting severe drifting, but after tuning the covariance and noise parameters in the info file, I've been able to get semi-usable results. Two problems remain and I was hoping to get your feedback though! 

1) The Z-height is inverted. Not sure how to approach that.
2) The system experiences severe position miscalculation on either the X or Y axes whenever yaw is introduced, but otherwise returns to normal when the camera is turned back to the initial yaw angle. Could it be a lack of synchronization?

The system right now is very stable when it keeps view of one scene. Only trouble is with yaw really. Below is my info file.

[firefly.txt](https://github.com/ethz-asl/rovio/files/2316555/firefly.txt)

bloesch, 2018-08-28:
I do feel both problems are related to something wrong with the camera-IMU extrinsics calibration. How did you do that?

indigomega, 2018-08-29:
I got it using a calibration from kalibr. I thought it was a pretty good calibration and that the extrinsics were pretty much on point. here is a picture of my setup

[image]

Razbotics, 2018-09-13:
Hey i have been following your project quite a while. Also i am working on similar autonomous drone project. Would u like to connect with me on linkdin?...i would like to discuss my experience with you

indigomega, 2018-09-13:
Hey, sure thing I’m always happy to help, just comment your question on my latest video. I would like other people to make use of our conversation for their own projects so discussing publicly is the way to go.

closed 2018-09-13 by indigomega