ethz-asl/rovio #200: Configuration for Odroid XU4

odinhr, 2018-11-19:
Hi all!

I'm testing now Rovio for Odroid and my output is about 2 Hz. With the same configuration on PC I get about 20-25 Hz. Can anyone says me a better configuration for Odroid?

Thank you.

Odín.

ZacharyTaylor, 2018-11-19:
Getting real time on an Odroid is going to be hard. The thing that will make the biggest difference is to go into the cmake file and drop the number of tracked features from 25 to something like 10.

chutsu, 2018-12-01:
You didn't mention anything about calibration. Calibration is vital for VIO to work. Would highly recommend you use Kalibr to obtain the calibration between camera and imu as well as camera intrinsics.

Further, the fact you're using ethernet to pass data around incurs time delays, if I'm right I don't think ROVIO fixes the time delay between images and imu so that could be another source of errors. In summary, I would say you have to synchronize your camera and imu properly in order do VIO, I believe this applies to all SLAM / VIO algorithms unless they explicitly estimate time delays as well. So in short your current hardware setup is not ideal for ROVIO in my opinion.

odinhr, 2018-12-03:
Hi @chutsu,

I've calibrated my configuration with kalibr. (View attached document) I've made a socket for "solving" the time delay which the MutlipleMachines ROS util could create via MAVROS.

For time synchronizing, I've tried with --time-calibration flag of Kalibr and with the ROS message filters function, but it can't improve. In my personal laptop, it works well sometimes, but I think it doesn't work on odroid  with the same configuration due to bad computation power. Maybe someone has a low computation cost configuration. Any help is welcome.

Sorry for my english and thank you so much.

odinhr, 2018-12-03:
[rovio.txt](https://github.com/ethz-asl/rovio/files/2638958/rovio.txt)