ethz-asl/rovio #201: Error when use camera of ar drone parrot 2

chekakta, 2018-11-29:
I have a problem (cv_bridge exception) when use the Rovio with Ar drone camera as an input (/ardrone/image_raw).
Any help?

[ERROR] [1543498947.803930866]: cv_bridge exception: [rgb8] is a color format but [8UC1] is not so they must have the same OpenCV type, CV_8UC3, CV16UC1 ....

odinhr, 2018-12-03:
Hi @chekakta,

I think your error is caused by the color format. You should give Rovio a grayscale image. You can convert your image to grayscale with the image_proc ROS node (http://wiki.ros.org/image_proc). Then subscribe Rovio to one of the new published topics(image_rect or image_mono). Try it!

closed 2019-02-17 by bloesch

RigerLee, 2019-08-11:
In my case, change sensor_msgs::image_encodings::TYPE_8UC1 to sensor_msgs::image_encodings::MONO8  in imgCallback solved the problem.