ethz-asl/rovio #202: How to add GPS?

str-ygomi, 2018-12-18:
Apologies in advance for beginner questions...

I am trying to run ROVIO using some low quality sensors. I want to improve the results by using some GPS data, to try and stop the divergence which happens about a minute into my recording. All of my data is just in txt and jpg format, so I've been using the Kalibr bag creator to make a bag file first, but now I am trying to use python to create the bag files so I can include GPS. What I don't really understand is how I should be including the GPS data; does the topic need to have a particular name? I see that there are optional sensors with names like GPS_WGS and GPS_UTC, does this mean the topic would contain GPS readouts? I tried just using some generic name like gps0, then using the "import_gps_data_from_rosbag" inside Maplab just as a check to see if it can be read but always just get a core dump.

I feel like I have tried to read any websites that contain anything like "GPS", "Maplab", "Rovio" etc but I still don't understand :( Any advice would be great!

Thanks in advance!

marco-tranzatto, 2018-12-19:
Hi,

I suggest you have a look here: [mav_rtk_gps](https://github.com/ethz-asl/mav_rtk_gps/wiki)
The wiki explains how we integrated the usage of GPS in Rovio.

Regards,
Marco.

str-ygomi, 2018-12-19:
Hi Marco,

Thanks for the response! I've taken a look at the wiki you mentioned, but I think I still have the same problem that all my GPS data is just in txt files, and I don't know how to get it to work properly as a ROS message. From what you posted, I got as far as the "RTK & Rovio..." GUI, but get an error about "No external IMU messages received" - I guess this is about the same problem ie I am not actually recording live data but trying to use prerecorded data in txt and jpegs. Do you have any tips on where I can go to figure this out?

Thanks!

marco-tranzatto, 2018-12-19:
Hi,

I think you should first convert your text file into ROS messages, and then use the existing pipeline.

Regards,
Marco.