ethz-asl/rovio #210: What is the safeWarningTime_

0Jiahao, 2019-03-20:
Hi! I am currently using [ethz-asl/realsense/devel/d435i](https://github.com/ethz-asl/realsense/tree/devel/d435i) with realsense D435i. I firstly recorded a rosbag for kalibr calibration and received the following result
`Calibration results
===================
Reprojection error squarred (cam0):  mean 0.0160805955519, median 0.00884699898598, std: 0.0230406098339
Reprojection error squarred (cam1):  mean 0.0141570452058, median 0.00737732199235, std: 0.0222017588545
Gyro error squarred (imu0):          mean 0.215724817809, median 0.00620870996076, std: 20.2846267125
Accelerometer error squarred (imu0): mean 0.105145329579, median 0.0409864492831, std: 0.303328080234
`
I think the calibration result is pretty nice. However, when I run the ROVIO with [ethz-asl/realsense/devel/d435i](https://github.com/ethz-asl/realsense/tree/devel/d435i), it keep warning me this message:
`[FilterBase::addUpdateMeas] Warning: included measurements at time (t) before safeTime (safeWarningTime_)`. I can see the `t` is growing while `safeWarningTime_` never changed. 

Any hints what happened or what the `safeWarningTime_` is? Thanks:)

beduffy, 2019-08-12:
I have this issue too, if anyone knows how to fix it, that would be much appreciated!

beduffy, 2019-08-12:
I fixed it by making sure that all my timestamps were before the initial reset time

0Jiahao, 2023-05-15:
seems there is a solution

closed 2023-05-15 by 0Jiahao