ethz-asl/rovio #213: I am a newbie, I want to run ROVIO on RealSenseD435i, what files should I modify?

VicentFan, 2019-04-11:


m0oz, 2019-05-10:
Make sure to check this repos wiki: https://github.com/ethz-asl/rovio/wiki
Essentially you will have to

- modify the config file rovio.info (see wiki for more details on which values to modify)

- create a new camera calibration .yaml with the intrinsics of your realsense. You can just copy one of the existing camera.yaml files and replace the intinsic values with the ones from your camera. Launch the rosnode of the realsense and echo the /camera_info topic to obtain the intrinsic parameters
- modify the params rovio_node.launch to point to the correct config and camera calibration files

SebastienFerrero, 2019-05-17:
Hello every one , 
Where can we find this VI-sensor? Is it available on the market?
[image]

Thanks
Sebastien