ethz-asl/rovio #214: where is the result?

EthanFresnel, 2019-04-15:
I have prepare all the depends and run the algorithm using the dataset EuRoc my dataset from mynt_eye. But I got a question that where is the test result? plz and thank u every much.

mechaliomar, 2019-05-14:
@EthanFresnel  I'm also using MYNTEYE camera, could you please tell me how to start from the beginning ? I had already installed rovio.

THANKS
BEST REGARDS

EthanFresnel, 2019-05-14:
According to my experience, you need do calibration firstly, and then
record your data by ROS. If you have your own data recorded by MYNTEYE, you
just need to connect the topic from the MYNTEYE.bag.

mechaliomar, 2019-05-14:
@ethan thanks for your reply,  so kind.  I'm planning to use mynteye and rovio for real time pose estimate.   As the uav is moving rovio estimate the 6 DOF, .  Or haven't you done this before? 

Best regards

EthanFresnel, 2019-05-14:
yep, I put on some sensors on a uva and just recorded data on a mini PC.
sensors are camera, imu, gps and some others. if u have problem on
calibration, u can use the calibration data from VINS-FUSION,
https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/tree/master/config/mynteye

this way save much time because the calibration always costs lots of time.
and then u can record data, run rovio, estimate algorithm.
I put the camera under uva and shoot the ground, but I don't think this
method is suitable for rovio, or it could be about the parameters. maybe u
can have a try.

mechaliomar, 2019-05-14:
@EthanFresnel xiexie, ni zhen hao,  ni hui shuo zhongwen ma? 
But why you say record data then run rovio?  Can't use rovio while the uav is moving? 
Thanks

EthanFresnel, 2019-05-14:
rovio is based on kalman filter, there's still a lot of computation. and I
just put a mini PC without high computing power. So I need to run rovio on
another common computer. if u have a better mini pc on uva, u really can
use rovio while moving.
才发现你上面写的是拼音……

mechaliomar, 2019-05-15:
@EthanFresnel  THANKS for your precise answer,yes EKF can be used in many real-time application as estimator.
What do you think about NVIDIA JESTON TX2 as embedded system to be used on the UAV, I'm planning to use it as I used it before for an implementation of real-time  path planning.

best regards
UESTC
CHINA

EthanFresnel, 2019-05-15:
I've only heard of NVIDIA JESTON TX2, but I haven't used it. I looked at
its parameters and thought it could be implemented. You can test it with a
public dataset firstly. And EuRoc dataset is a good choice.

Sincerely

Ethan (Master)
Research Center of Deep Space Robot.
Department of Electronic and Control Engineering Beijing University of
Technology
100 Ping Le Yuan, Chaoyang District, Beijing 100124, CHINA

mechaliomar, 2019-05-15:
@EthanFresnel  my email is mechaliomar@std.uestc.edu.cn 
we can exchange our wechat.

best regards

mechaliomar, 2019-05-15:
@EthanFresnel  can  you show me how to run the dataset for rovio, I tried a lot but....

thanks