ethz-asl/rovio #216: Can rovio be used in real time 6DOF pose estimate?

mechaliomar, 2019-05-14:
Hello, 🙂
Thanks for this great code. 
I would like to work on path planning in real time,  so I need real time 6DOF pose estimate for the uav. 
-So,  I would like to know if this cute tool 'rovio' could be used for real time estimate ( as the uav is moving), 
-In which frame does Mr 'rovio' give us the pose estimate? 

Thanks a lot
Best regards

Jaeyoung-Lim, 2019-05-15:
@mechaliomar Have you gone through the papers available in the readme?

michaelpantic, 2019-05-15:
a) Yes.
b) Per default, the frame is initialized in the position you start rovio,

Please read the papers, wikis and readmes. There is a lot of material about visual-inertial state estimation out there.
You will need a good IMU that is hardware synchronized to a global shutter camera.

closed 2019-05-15 by michaelpantic