ethz-asl/rovio #218: WHERE to modify there  different build variables?

mechaliomar, 2019-05-15:
HI

Build Configuration
There are different build variables that can be set when compiling rovio:
ROVIO_NMAXFEATURE: Maximal number of features for rovio
....

THANKS A LOT

michaelpantic, 2019-05-15:
Please read the wiki before opening issues for questions,  as these things are documented: 
https://github.com/ethz-asl/rovio/wiki/Configuration

michaelpantic, 2019-05-15:
Also there wasn't even a question in your issue :-)

closed 2019-05-15 by michaelpantic

mechaliomar, 2019-05-15:
@michaelpantic I have been reading very carefully your wiki, and what's the problem if I didn't understand,  we are not all experts. Maybe it's clear for someone but it's not clear for me. 
I didn't find the file in which I modify the values. 
Please do not Prejudge people in wrong way!

helenol, 2019-05-15:
Hi @mechaliomar, if you're struggling to understand how a package is compiled, I'm not sure I believe you've "closely read" the ROS tutorials.
Please check out these other resources: http://wiki.ros.org/ROS/Tutorials/catkin/CreatingPackage and http://wiki.ros.org/catkin/CMakeLists.txt

While I understand your frustration and some of this stuff is tough to get started with, *please* do a google search, read some tutorials, or try to solve the issue yourself before asking questions.
If you genuinely feel you've done this as much as you can before you've asked the issues, I'd be happy to refer you to some of the previous issues you've posted and how many can be resolved with a google search, reading the abstract of the relevant papers, or a ROS tutorial.

We're not trying to be jerks, but please understand that we *also* have jobs to do that do not involve hand-holding your way through learning ROS, robotics, or your PhD.

mechaliomar, 2019-05-15:
THANKS. But believe me I'm not searching a prepacked piece of cake. Maybe my question are very basic, I understand, But Mrs Helenol be sure that I have moral,and I know that's difficult for you to deal with many things. Also, I hardly try understand before I ask you, may be I'm like Einstein, they said he had troubles with fast understanding lol ... hhh

Sorry if I hurt the feeling of your team, 
BEST REGARDS