ethz-asl/rovio #225: Model jacInput Test failed

JiatianWu, 2019-07-24:
When I launch rrovio_node and I got below:
```
Testing Prediction
==== Test successful (6.72468e-08) ====
==== Test successful (4.9395e-08) ====
Testing cameraOutputCF
==== Test successful (4.44089e-08) ====
Testing imuOutputCF
==== Test successful (1.03452e-08) ====
Testing attitudeToYprCF
==== Test successful (4.6604e-08) ====
Testing transformFeatureOutputCT
==== Test successful (1.0206e-08) ====
Testing LandmarkOutputImuCT
==== Test successful (3.57083e-08) ====
Testing FeatureOutputReadableCT
==== Test successful (1.21661e-08) ====
Testing pixelOutputCT (can sometimes exhibit large absolut errors due to the float precision)
==== Model jacInput Test failed: 74.0615 is larger than 1 at row 1(def_0) and col 0() ====
  -72871.9  -72797.9
Testing zero velocity update
==== Test successful (3.04311e-10) ====
==== Test successful (1.39778e-10) ====
```
Does anyone know why the model jacInput test failed? It looks like it will results in large absolute errors.

AnfangRobkit, 2020-01-14:
Have you solved this problem? I also meet the same error

JiatianWu, 2020-01-19:
Yeah. Initialize `n` in `test finite difference Jacobians` of `lwf/testPrediction.cpp`

````
TYPED_TEST(PredictionModelTest, FDjacobians) {
  Eigen::MatrixXd F((int)(TestFixture::mtPredictionExample::mtState::D_),(int)(TestFixture::mtPredictionExample::mtState::D_));
  Eigen::MatrixXd F_FD((int)(TestFixture::mtPredictionExample::mtState::D_),(int)(TestFixture::mtPredictionExample::mtState::D_));
  typename TestFixture::mtPredictionNoise n;
  n.setIdentity();
  this->testPrediction_.meas_ = this->testPredictionMeas_;
  this->testPrediction_.template jacInputFD<0>(F_FD,std::forward_as_tuple(this->testState_,n),this->dt_,0.0000001);
  this->testPrediction_.template jacInput<0>(F,std::forward_as_tuple(this->testState_,n),this->dt_);
  Eigen::MatrixXd Fn((int)(TestFixture::mtPredictionExample::mtState::D_),(int)(TestFixture::mtPredictionExample::mtNoise::D_));
  Eigen::MatrixXd Fn_FD((int)(TestFixture::mtPredictionExample::mtState::D_),(int)(TestFixture::mtPredictionExample::mtNoise::D_));
  this->testPrediction_.template jacInputFD<1>(Fn_FD,std::forward_as_tuple(this->testState_,n),this->dt_,0.0000001);
  this->testPrediction_.template jacInput<1>(Fn,std::forward_as_tuple(this->testState_,n),this->dt_);
  switch(TestFixture::id_){
    case 0:
      ASSERT_NEAR((F-F_FD).norm(),0.0,1e-5);
      ASSERT_NEAR((Fn-Fn_FD).norm(),0.0,1e-5);
      break;
    case 1:
      ASSERT_NEAR((F-F_FD).norm(),0.0,1e-8);
      ASSERT_NEAR((Fn-Fn_FD).norm(),0.0,1e-8);
      break;
    default:
      ASSERT_NEAR((F-F_FD).norm(),0.0,1e-5);
      ASSERT_NEAR((Fn-Fn_FD).norm(),0.0,1e-5);
  };
}

````