ethz-asl/rovio #226: Rovio doesnt work in virtual scene with ideal conditions

Mikhail2017, 2019-09-25:
Hi all.
I have created virtual scene and calculated high precision positions and imu measurements. Of course imu and camera frames are absolutely synchronized. 
When running rovio I've got a lot of REJECTED (iterations did no converge) messages and serious problems with update stage perfomance.
EurocMav dataset works well.
Please help!

Here are my configs for virtual scene.
[carla_cfgs.zip](https://github.com/ethz-asl/rovio/files/3652588/carla_cfgs.zip)

Mikhail2017, 2019-09-25:
self checks output:

Testing Prediction
==== Model jacInput Test failed: 1 is larger than 1e-06 at row 15(vep) and col 15(vep) ====
  1  0
==== Test successful (6.45948e-08) ====
Testing cameraOutputCF
==== Test successful (1.03452e-08) ====
Testing imuOutputCF
==== Test successful (1.03452e-08) ====
Testing attitudeToYprCF
==== Test successful (4.6604e-08) ====
Testing transformFeatureOutputCT
==== Test successful (1.0206e-08) ====
Testing LandmarkOutputImuCT
==== Test successful (7.9124e-09) ====
Testing FeatureOutputReadableCT
==== Test successful (1.21661e-08) ====
Testing pixelOutputCT (can sometimes exhibit large absolut errors due to the float precision)
==== Model jacInput Test failed: 2.78337 is larger than 1 at row 0(def_0) and col 0() ====
  7323.34  7320.56
Testing zero velocity update
==== Test successful (3.04311e-10) ====
==== Test successful (1.39778e-10) ====

Mikhail2017, 2019-09-25:
I've tried to setup zero covariances for imu in config file, unfortunately it doesnt affect result

Mikhail2017, 2019-09-25:
Tried to setup imu axis like in Euroc MAV also without any success. Still divergence

Mikhail2017, 2019-09-26:
I've realized that the field VelocityUpdate was missingin my config. That was the only reason for poor work.

closed 2019-09-26 by Mikhail2017

ItsTimmy, 2020-04-24:
@Mikhail2017 I am also trying to run Rovio with the Carla simulator. I was wondering if you could help me out? I have a few questions:

 - What kind of accuracy did you get?
 - I am currently having a problem where the covariance of Rovio's pose estimate increases very quickly (~10 to 100 per second), until Rovio completely diverges after 1 to 2 minutes. Did you ever experience this problem?
 - Did you have to set any specific settings, in either Carla or Rovio, to get it to work?
 - Do you have any data recorded from Carla that I could experiment with? Perhaps a ROS bag?

Thank you for any help you can provide.