ethz-asl/rovio #228: extrinsics parameters from euroc dataset and rovio.info are different

AnfangRobkit, 2019-12-27:
Hi,

I noticed that extrinsic parameters from euroc dataset and rovio.info are different, of course, I have transferred the rotation matrix into the quaternion format before the comparison,
```
quaternion drawn from rovio.info \[0.00666398307551 -0.0079168224269 -0.701985972528 0.712115587266\]
quaternion from euroc dataset converted from rotation matrix is 0.712 -0.008i +0.010j +0.702k
```

Are the differences above are acceptable?

And in my view, the translation vector should be kind of same, since they are independent of the expression format of the rotation, however,
the vector in euroc:
```
[-0.021640145497, -0.064676986768, 0.009810730590\]
```
the vecor in rovio.info:
```
 -0.011167419918 -0.0574640920022 0.0207586947896
```
of course, these two set of parameters could get some kind of odometry results when online-calibration is on, but could you share some hints w.r.t these parameters?

smauq, 2020-01-01:
Did you take into account that in the camera yaml it's T_BS (sensor to base) while in the rovio info it's qCM (imu to camera) which is the inverse. This also accounts for why the translation is different since it's expressed in a different frame.

AnfangRobkit, 2020-01-08:
Hi, thanks for your reply and helpful information.

I did not notice that that should be T_BS, since I have assumed that they all follow the ASL Kalibr format, where they only use T_cam_imu to transform from IMU to camera coordinates as [link.](https://github.com/ethz-asl/kalibr/wiki/yaml-formats)

Based on your information, I created one python file, which trys to turn the  T_BS to qCM, as follows:
```python
from pyquaternion import Quaternion
import numpy as np
T_bs = np.matrix([[0.0148655429818, -0.999880929698, 0.00414029679422, -0.0216401454975],
         [0.999557249008, 0.0149672133247, 0.025715529948, -0.064676986768],
        [-0.0257744366974, 0.00375618835797, 0.999660727178, 0.00981073058949],
         [0.0, 0.0, 0.0, 1.0]])
T_sb = np.linalg.inv(T_bs)
quat_euro_ = Quaternion(matrix=T_sb[0:3, 0:3])
```
it turns out to be better result than before, since qCM from ROVIO config is 0.712 + 0.0067i - 0.008j -0.702k, and my calculation result is 0.712 + 0.0077i -0.010j -0.702k, the slight difference here I guess is acceptable now.

However, I am not quite sure about the translation vector, if I directly taken the first three elements of the last column from the inversed transformation matrix, it should be [0.065, -0.207, -0.008], which is not quite exactly as described in ROVIO.info.

Thx in advence! :)

tvandenzegel, 2025-02-03:
Did somebody eventually figure out how to transform the translation vector from euroc dataset to rovio format? 
Because i have the same question as @SiyuanHuang95 about the translation

specarmi, 2025-02-16:
The values in `rovio.info` are definitely wrong. Ignoring reference frames the norm of the IMU to camera translation (and vice versa) is `0.06890325790004832` according to the EuRoC calibration results. The norm of the same translation from `rovio.info` is `0.062110840814189994`. The norm should always be the same (doesn't matter what frame the translation is expressed in) so `rovio.info` is wrong. My only guess is that they were purposefully evaluating the ability to calibrate the extrinsics online and forgot to change it back.

Here is my best attempt at getting the right values for the rovio.info file (for Camera0). Note that the translation should **not** be taken from the `T_imu_to_camera` matrix (as was done in the code snippet above). 
```
import numpy as np
from scipy.spatial.transform import Rotation

T_camera_to_imu = np.array([
    [0.0148655429818, -0.999880929698, 0.00414029679422, -0.0216401454975],
    [0.999557249008, 0.0149672133247, 0.025715529948, -0.064676986768],
    [-0.0257744366974, 0.00375618835797, 0.999660727178, 0.00981073058949],
    [0.0, 0.0, 0.0, 1.0]])
T_imu_to_camera = np.linalg.inv(T_camera_to_imu)

rovio_translation = T_camera_to_imu[:3, 3]

R_imu_to_camera = T_imu_to_camera[:3, :3]
rovio_quat = Rotation.from_matrix(R_imu_to_camera).as_quat()

print(rovio_translation)
print(rovio_quat)
```

This yields
```
[-0.02164015 -0.06467699  0.00981073]
[ 0.00770718 -0.01049932 -0.7017528   0.71230146]
```